chipweinberger/flutter_blue_plus #432: .discoverDevices() throws [CoreBluetooth] API MISUSE: can only accept commands while in the connected state

sudoadarsh, 2023-07-26:
Steps to reproduce:

1. Connect to a Bluetooth Device.
2. Turn off Bluetooth adapter.
3. Connect to the Bluetooth Device and listen to device.connectionState
4. When connectionState == BluetoothConnectionState.connected, use the device BluetoothDevice Object to call .discoverDevices()

chipweinberger, 2023-07-26:
discoverDevices? That does not exist...

sudoadarsh, 2023-07-26:
The method is .discoverServices() and for the logs, I have reverted my code back to plugin version 1.4.0 and it's working fine.

chipweinberger, 2023-07-26:
Yes, the logs would be very helpful. All of them.

chipweinberger, 2023-07-26:
Can you explain these steps more?

2. Turn off Bluetooth adapter.
3. Connect to the Bluetooth Device and listen to device.connectionState

How are yo connecting while the adapter is off?

chipweinberger, 2023-07-26:
should be fixed in 1.11.5

closed 2023-07-26 by chipweinberger