chipweinberger/flutter_blue_plus #434: [Bug]: Location not working after Flutter Blue Plus package update to version 1.7.0 and above

spehj, 2023-07-26:
### FlutterBluePlus Version

1.7.0

### What OS?

Android

### OS Version

Android 12

### What happened?

Since the Flutter Blue Plus package was updated to version 1.7.0, commit 8a35c6a0b6aca2fe23e5cd3730b7d3cf7903437c, we have encountered an issue with the location functionality in our app. We use the 'location' package (v4.4.0) for handling location services. The app doesn't even ask for location permission when started. Permissions are requested like this (permisson_handler package v10.4.3) :

```dart
Map<Permission, PermissionStatus> statuses = await [
        Permission.notification,
        Permission.bluetooth,
        Permission.bluetoothAdvertise,
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        Permission.locationWhenInUse,
        Permission.location,
      ].request();
```
Our AndroidManifest is like this:

```xml
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
```

In the mentioned commit, the changes were made to the AndroidManifest.xml file of the package as follows:

Before the update:

```xml
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.boskokg.flutter_blue_plus">

    <!-- New Bluetooth permissions in Android 12
           https://developer.android.com/about/versions/12/features/bluetooth-permissions
      -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
</manifest>
```

After update: 
```xml
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.boskokg.flutter_blue_plus">

  <!-- New Bluetooth permissions in Android 12
          https://developer.android.com/about/versions/12/features/bluetooth-permissions
    -->
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

  <!-- legacy for Android 11 or lower -->
  <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />

  <!-- legacy for Android 9 or lower -->
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
</manifest>
```

We suspect that adding android:maxSdkVersion to android.permission.ACCESS_FINE_LOCATION in the new Bluetooth permissions is causing conflicts with the 'location' package. This could be why the location functionality is no longer working after the update.

To resolve this issue we suggest removing android:maxSdkVersion from android.permission.ACCESS_FINE_LOCATION in Flutter Blue Plus AndroidManifest.

This issue was reproduced on Samsung Galaxy S10e (Android 12), and Google Pixel 7 Pro (Android 13).

### Logs

```shell
No logs.
```

chipweinberger, 2023-07-26:
Did you try this?

```
<!-- remove the plugin permission which sets neverForLocation -->
<uses-permission
    android:name="android.permission.BLUETOOTH_SCAN"
    tools:node="remove"
    tools:selector="com.boskokg.flutter_blue_plus">
</uses-permission>

<!-- add back without neverForLocation -->
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
```

spehj, 2023-07-26:
Thanks @chipweinberger  and @krikristoophe. I added your suggested code to our project's AndroidManifest, but it doesn't help. The problem is still in ACCESS_FINE_LOCATION in flutter blue plus AndroidManifest. 

Do you have any other suggestions how to keep the 

```xml
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
```

in Flutter Blue Plus AndroidManifest, but override it in our project's AndroidManifest to still request ACCESS_FINE_LOCATION permission?

krikristoophe, 2023-07-26:
@spehj you can also try to add 

```
<uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove"
        tools:selector="com.boskokg.flutter_blue_plus">
    </uses-permission>
```

I just test it on my project and it works

closed 2023-07-26 by chipweinberger

zigapovhe, 2023-11-20:
If anyone is wondering, after v1.16.5 you need to use _**com.lib.flutter_blue_plus**_ in tools:selector.

chipweinberger, 2023-11-20:
that's true, but we also removed all permissions from flutter blue plus, so you should not need that anymore.