chipweinberger/flutter_blue_plus #436: [Question]: await FlutterBluePlus.isAvailable is false on iOS 16.6 even bluetooth is activated

parrotdance, 2023-07-26:
### Ask your question

As tutorial described, I have:
- Added permissions for iOS in `Info.plist` file.
- Confirmed the bluetooth request
- Bluetooth and Local Network is enabled in Settings

But error still threw from the code below:
```dart
if (await FlutterBluePlus.isAvailable == false) {
  throw 'Bluetooth not supported by this device';
}
```

I have checked `FlutterBluePlusPlugin.m` and have no clues about it. Is there anything I missed?

chipweinberger, 2023-07-26:
You could try calling `await FlutterBluePlus.adapterState` to see what the actual state is.

chipweinberger, 2023-07-26:
Nevermind `adapterState` wont work without an adapter. I'm pushing an update to fix `adapterState`.

chipweinberger, 2023-07-26:
try calling `await FlutterBluePlus.adapterState.first` on 1.11.8

parrotdance, 2023-07-27:
@chipweinberger Thanks bro, in fact I have tried to print the value of `FlutterBluePlus.adapterState` before I wrote this issue, but I forgot the result.

Today when I start the app, Flutter reported an error about sort of application signature. I build the iOS project in Xcode and start my app again form flutter, the error not appeared and bluetooth works as expected.

I am not good at iOS development so I can't confirm if I must re-build project after update `Info.plist` file to make it work.

However, this issue is solved and it's time to closed. Thanks for your help again.

closed 2023-07-27 by parrotdance