chipweinberger/flutter_blue_plus #440: [Bug]:

joyhope, 2023-07-27:
### FlutterBluePlus Version

1.12.4

### What OS?

Android

### OS Version

android 12

### What happened?

Unable to find BLE device.   1.11.X version is ok.   No permisssion indicate, 

### Logs

```shell
07-27 10:55:00.669 12527 12580 E flutter : [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, FlutterBluePlus requires android.permission.BLUETOOTH_SCAN permission, null, null)
07-27 10:55:00.669 12527 12580 E flutter : #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652)
07-27 10:55:00.669 12527 12580 E flutter : #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310)
07-27 10:55:00.669 12527 12580 E flutter : <asynchronous suspension>
07-27 10:55:00.669 12527 12580 E flutter : #2      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:178)
07-27 10:55:00.669 12527 12580 E flutter : <asynchronous suspension>
07-27 10:55:00.669 12527 12580 E flutter : #3      Isolate.run.<anonymous closure> (dart:isolate:291)
07-27 10:55:00.669 12527 12580 E flutter : <asynchronous suspension>
07-27 10:55:00.669 12527 12580 E flutter :
```

joyhope, 2023-07-27:
sorry,  my phone is HarmonyOS 3.0 ( It may be different from Android 12)

joyhope, 2023-07-27:
Seems manifest file is missing.

shubhamsinghmutualmobile, 2023-07-27:
See its solution here: https://github.com/boskokg/flutter_blue_plus/issues/409#issuecomment-1651055224

joyhope, 2023-07-27:
so user's own app needs to define this permssion.

joyhope, 2023-07-27:
Permission was added in the user's app's manifest, and it works.

shubhamsinghmutualmobile, 2023-07-27:
> so user's own app needs to define this permission.

Apparently, on Android 12 and above, we don't require the Location permission in order to just scan for devices, that's what the `BLUETOOTH_SCAN` permission is for and to be on a privacy-safer side, the package maintainers have chosen to not use the location permission for A12 and above that you can easily override with the method shared above.

shubhamsinghmutualmobile, 2023-07-27:
> Permission was added in the user's app's manifest, and it works.

@joyhope If there are any other questions regarding this issue, you can drop them here and feel free to close the issue whenever you feel it is resolved for you 👍

chipweinberger, 2023-07-27:
> Permission was added in the user's app's manifest, and it works.

Yes you need to specify permissions yourself.

closed 2023-07-27 by joyhope