chipweinberger/flutter_blue_plus #441: [Bug]: scan not find any devices

WeiKuang, 2023-07-27:
### FlutterBluePlus Version

1.12.4

### What OS?

iOS

### OS Version

16.4.1

### What happened?

Same code, different performance on two versions.
only happens on scan.
in 1.10.10, it is ok.
but in 1.12.4, not find any devices.
```dart
var subscription = FlutterBluePlus.scanResults.listen((results) {
    for (ScanResult r in results) {
        Map<int, List<int>> map = r.advertisementData.manufacturerData;
    }
});

// Start scanning
FlutterBluePlus.startScan(timeout: Duration(seconds: 4));
```

### Logs

```shell
version: 1.10.10
[FBP-iOS] handleMethodCall: isOn
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: getConnectedDevices
[FBP-iOS] handleMethodCall: startScan
version: 1.12.4
[FBP-iOS] handleMethodCall: isOn
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: startScan
```

chipweinberger, 2023-07-27:
works for me (macOS)

WeiKuang, 2023-07-27:
Of course no problem, glad to help. 
But I'm sure I can fix it because I'm a new coder.

chipweinberger, 2023-07-27:
FlutterBluePlus.logLevel to verbose

WeiKuang, 2023-07-27:
it seems to be stuck in a loop。
[FBP-iOS] handleMethodCall: isOn
flutter: \^[[38;5;12m┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
flutter: \^[[38;5;12m│ 16:30:14.832 (+0:00:24.104448)<…>
flutter: \^[[38;5;12m├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄<…>
flutter: \^[[38;5;12m│ 💡 _scan Filter Device<…>
flutter: \^[[38;5;12m└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
[FBP-iOS] handleMethodCall: stopScan
flutter: 
BmScanSettings {service_uuids: [], mac_addresses: [], allow_duplicates: false, android_scan_mode: 2, android_uses_fine_location: false}
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral

chipweinberger, 2023-07-27:
It is not a loop, there are just a lot of devices, and maybe you have allowDuplicates turned on.

I just tried the example app and it works for me on 1.12.4, with iOS 16.5.1

chipweinberger, 2023-07-27:
You should put a breakpoint here

[image]

WeiKuang, 2023-07-27:
Oh, it's my fault.
my code is :
try {
  var subscription = FlutterBluePlus.scanResults.listen((results) {
    for (ScanResult r in results) {
      Map<int, List<int>> map = r.advertisementData.manufacturerData;
    }
  });
  FlutterBluePlus.startScan(timeout: const Duration(seconds: 5));
} catch (e) {
  LogUtils.logger!.e('Error during Bluetooth scan: $e');
} finally {
  FlutterBluePlus.stopScan();
}
When the scan starts, it stops immediately。
so remove finally is OK

closed 2023-07-27 by WeiKuang

chipweinberger, 2023-07-27:
great!