chipweinberger/flutter_blue_plus #442: [Bug]: PlatformException - ERROR_GATT_WRITE_REQUEST_BUSY

FritzMatthaeus, 2023-07-27:
### FlutterBluePlus Version

1.12.4

### What OS?

Android

### OS Version

Android 13

### What happened?

The device is connect and i can succesfully try to read a characteristic. But if i try to write to a characteristic

```
characteristic?.write(bytes, withoutResponse: true);
```
it fails. I cannot see any error in the logs
```
D/[FBP-Android](21900): [FBP-Android] onMethodCall: writeCharacteristic
D/TrafficStats(21900): tagSocket(157) with statsTag=0xffffffff, statsUid=-1
```
but i do find this Exception on Sentry ( see logs for further details)
```
PlatformException(write_characteristic_error, gatt.writeCharacteristic() returned 201 : ERROR_GATT_WRITE_REQUEST_BUSY, null, null)
```

I do not have this problem in a FlutterBluePlus Version 1.7.5 which was the latest Version i have been using.

### Logs

```shell
PlatformException: PlatformException(write_characteristic_error, gatt.writeCharacteristic() returned 201 : ERROR_GATT_WRITE_REQUEST_BUSY, null, null)
  File "message_codecs.dart", line 652, col 7, in StandardMethodCodec.decodeEnvelope
  File "platform_channel.dart", line 310, col 18, in MethodChannel._invokeMethod
  File "<asynchronous suspension>"
  File "bluetooth_characteristic.dart", line 142, col 7, in BluetoothCharacteristic.write.<fn>
  File "<asynchronous suspension>"
  File "utils.dart", line 371, col 7, in _Mutex.synchronized
  File "<asynchronous suspension>"
  File "bluetooth_characteristic.dart", line 119, col 5, in BluetoothCharacteristic.write
```

FritzMatthaeus, 2023-07-27:
This was a misleading report, please ignore.

closed 2023-07-27 by FritzMatthaeus

Roee-Tsur, 2023-08-08:
@FritzMatthaeus why was this misleading?
What was your actual problem and how did you solve it?

I'm also getting this exception.

FritzMatthaeus, 2023-08-08:
@Roee-Tsur 

my issue was misleading as this error was not causing my problem, as i assumed when posting it.

FritzMatthaeus, 2023-08-08:
My actual problem was that i could not write to a characteristic due to MtuSize restrictions. In earlier version there was no check on the size of the byte array to be send, but there is one now. That was causing my error.

chipweinberger, 2023-09-08:
The mtu size restriction is gone. use `chr.write(allowLongWrite:true)`

FritzMatthaeus, 2023-09-11:
Hi,

we've investigated this problem a bit further and found some irritating results. We have a specific ble-device we are connecting to. In the device logs we can see, that the negotiated MTU Size is the maximum corresponding to the platform (iOS, Android). But the MTU-value i get from the package is always 23. If i request another MTU size (on Android), this is successful in the device logs but i never get a successfull callback in the package. 

This is why my write (withoutResponse: true) is successful even if i send a larger byte array than the 23 defined by the MTU variable.

Any ideas on why i do not get a correct MTU size after the connection is established?

chipweinberger, 2023-09-11:
thanks for the follow up

i'd open a new issue with verbose logs enabled :)

chipweinberger, 2023-09-11:
also, if the problem is ios pleasure update fbp

mtu had as bug.