chipweinberger/flutter_blue_plus #445: [Feature]: Ignore MTU Size in write method

FritzMatthaeus, 2023-07-27:
### Feature Request

For communication to a special device i need to send larger byte arrays than the negotiated MTU size allows. Therefore i added a flag to overwrite this MTU size check. A pull request has been made

chipweinberger, 2023-07-27:
it will fail on android.

android does not support this.

you need to split up your data it's very simple!

chipweinberger, 2023-07-27:
The reason this works on IOS is that apple will split up the message for you into MTU sized chunks.

So you think you are writing bigger packets, but you are not actually. 

So how will you circumvent that?

chipweinberger, 2023-07-28:
And this is what android says:

"When performing a write request operation (write without response), the data sent is truncated to the MTU size."

For writeWithResponse, it seems android will split up the message for you automatically (I'm not positive).

Perhaps different android versions and phones may behave differently. 

Have you confirmed that your "large" writes are actually being received by the system in large chunks? Because I dont think they are!

You should just split up the message yourself into MTU size chunks (23 in your case).

FritzMatthaeus, 2023-07-28:
I do agree with your reflections on this topic. My use case is a really special one and probably should not be respected in the package. 
---
I have been using the package for more than a year on this use case. It is (for whatever reason) not possible to successfully request a change in the MTU size, it's always 23. The hardware expects me to send data in a loop with 95 byte chunks. So i write 95 bytes in a write without response. 

I can see in the hardware logs, that the controller is receiving 95 bytes and that the Checksum is correct. We circumvent the problems of data loss you mentioned by having a workaround. The receiving controller is emitting a notification telling me to send the next package in case of success or re-sending the same package in case of Checksum errors.
---
This has been working fine on Android and iOS. So either we have been successfully ignoring the MTU size or iOS & Android did split up the data in 23 byte chunks and told the receiving bluetooth module to reconstruct the data. I don't know much about the hardware side of bluetooth.

We have tried to only send 23 bytes corresponding to the MTU size in each loop, but this slowed down the data transfer drastically. 
---

As i said, the package should probably not respect this special use case. But it did work properly without any problems on iOS and Android.

chipweinberger, 2023-07-28:
thanks for the follow up!

what you say is definitely interesting behavior and differs with the documentation I read online about iOS and Android splitting by mtu.

that is unfortunate that it is hurting data rate.

So iOS and Android ignore mtu and don't splitting according to your observations.

it would be nice to have clear answers about these behaviors. 

maybe there is some other explanation. 

this package will not have as many changes in the future, so it might be reasonable for you to maintain a fork.

chipweinberger, 2023-07-28:
similar to #448

chipweinberger, 2023-08-01:
closing. You are right that this should probably not be added as a feature. 

If you *need* this feature, make a clone of FBP and modify it yourself

---

```
cd /user/downloads
git clone https://github.com/boskokg/flutter_blue_plus.git
```

then in pubspec.yaml add the repo by path:

```
  flutter_blue_plus:
    path: /user/downloads/flutter_blue_plus
```
Now you can edit the FlutterBluePlus code.

closed 2023-08-01 by chipweinberger

chipweinberger, 2024-01-14:
btw, you can now use `allowLongWrite` to ignore mtu checks