chipweinberger/flutter_blue_plus #447: [Feature]: overall governance for plugin and API changes

ekuleshov, 2023-07-27:
### Feature Request

There been a lot of API changes and small updates in the plugin. It is great to see a very active development and it is most appreciated.

However the process is somewhat chaotic and it does affect all consumers/users of the plugin. Given that `flutter_blue_plus` plugin is the only one that provides support to work with BLE on Flutter, the plugin may need some kind of overall governance. Here are a few examples:

* Looking at the recent change log, there been a quite a lot of regression in the _release versions_ of plugin. For the end user that basically mean you don't know if you can update to the latest version or you are getting a time bomb with some regressions. To address that, I'd suggest to push changes into a `prereleases` instead, e.g. see https://pub.dev/packages/flutter_inappwebview/versions

* Public API changes. I understand the desire to provide a nice and clean API. However there are existing consumers of this plugin and all public API changes need to be weighted on and shouldn't be changed lightly every other day. Also, some modules depending on `flutter_blue_plus` provide additional functionality on top of the `flutter_blue_plus` and also exporting its API to their own consumers. Now consumers of these packages can't update either dependency and have to wait when both modules are updated.

* Other plugin changes specific to native platforms support. For example: "android: remove permissions from plugin. It is easier for user to specify everything".
With multiple Android versions any app has to support, the "user" has to know what permissions to put into their Android app manifests. _At very least that is not any "easier"_. More over - these permissions has to be in sync with functionality supported by the `flutter_blue_plus`. Basically a system module like `flutter_blue_plus` that provides access to native functionality on multiple target platforms should not offload (or leak) knowledge about functionality it supporting to the consumers or end users. The plugin should provide reasonable defaults to work out of the box and probably have an option to override these default (e.g. using standard Android mechanisms for merging manifests).
Requesting required permissions is also specific to each platform and even platform version (e.g. permissions are different between different versions of Android and iOS).

I can probably add a few more things to the list, but those are affecting me as a consumer of `flutter_blue_plus` the most. I hope these issues could be addressed for the long run.

chipweinberger, 2023-07-27:
Thanks for providing your input.

Just so you know, my changes to this library are basically done. 

My goal was to just add error handling, and fix all the open bugs.

There will be (should be) no more breaking changes going forward.

You are right that the `.instance` change really should have gone in `2.0`.

I think I'll probably leave the permissions as-is. It depends on whatever results in more issues filed haha. A lot of people opened issues not knowing how to merge manifests which was annoying. That said we did find a working solution to it in the end.

jfacoustic, 2023-07-27:
@chipweinberger Awesome to hear.   You've put in a ton of work over the last month and it's awesome!  What plans do you have on maintenance going forward?  Will you be available to review PR's for bugfixes/updates to the android API?

chipweinberger, 2023-07-27:
Not sure!

I've removed all the dependencies so that maintenance should be minimal!

CoreBluetooth and Android bluetooth are very stable at this point.

The goal is to have all the bugs fixed! heh.

but we'll see

closed 2023-07-28 by chipweinberger

chipweinberger, 2024-05-07:
[image]

just wanted to point out, commits & changes have really died down since this issue was filed, as I predicted. Once all the issues were fixed, there's nothing left to do really.

ekuleshov, 2024-05-07:
> just wanted to point out, commits & changes have really died down since this issue was filed.
> Once all the issues were fixed, there was nothing left to do.

This proposal had nothing to do with a rate of commit or changes. It was about what is being changed or why. Personally, I gave up reporting issues or commenting because no point.

This plugin APIs and internal decisions still requires app developers and plugin consumers deal with platform specifics, which should be handled by plugin out of the box. No roadmap or a glimpse of support for other platforms. And, with a single plugin maintainer, chances of this eventually die is pretty high.

chipweinberger, 2024-05-08:
thank you for your insights