chipweinberger/flutter_blue_plus #448: [Bug]: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null)

18723309899, 2023-07-28:
### FlutterBluePlus Version

flutter_blue_plus: ^1.12.7

### What OS?

Android

### OS Version

Android 33

### What happened?

When I was using 'write' to send data, 'withResponses' passed in' ture 'and reported an error. I know it may be because my device does not support this feature, but I still want to enable it because I can set it to' ture 'in' flutter_blue_plus: ^ 1.5.2 '. This way, my device can send data quickly and I don't need to worry about packet loss. The problem now is that I need to wait for the device to respond for a long time, So I hope you can change back to the original state and allow me to set it freely. Thank you!

### Logs

```shell
I/flutter ( 5278):PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null)
I/flutter ( 5278): withoutResponse true
```

chipweinberger, 2023-07-28:
very interesting. i'm surprised that android does not throw error.

i'll consider adding the option remove this check

chipweinberger, 2023-07-28:
are you sure that data rate is affected?

perhaps iOS and Android ignore this option and wait for response anyway

chipweinberger, 2023-07-28:
this is what ChatGPT says

"In iOS, the Core Bluetooth framework does check if a characteristic supports the specific write type (write with or without response) before attempting to write. 

If you try to write without response to a characteristic that does not support this operation, you'll receive a failure callback in the `peripheral:didWriteValueForCharacteristic:error:` delegate method. The error object will contain details about why the operation failed."

chipweinberger, 2023-07-28:
"On Android, the behavior is similar but the approach to handling it is slightly different. If you try to write without response to a characteristic that doesn't support it, the write operation will fail.

In Android, when interacting with Bluetooth Low Energy (BLE) devices via the Android Bluetooth API, you should always check the properties of the characteristic before attempting to write to it."

chipweinberger, 2023-07-28:
"For both iOS and Android, the BLE APIs provided by the system typically do not allow you to directly specify the opcode (like 0x52 or 0x12) when you're writing to a characteristic. Instead, you specify the type of write operation at a higher level (write with response or write without response), and the system takes care of mapping this to the appropriate opcode.

That being said, if you attempt to write without response (which maps to opcode 0x52) to a characteristic that only supports write with response (which maps to opcode 0x12), the system behavior will vary:

- On iOS, if you try to perform a write without response on a characteristic that doesn't support it, you will receive a failure callback in the `peripheral:didWriteValueForCharacteristic:error:` delegate method, indicating that the write operation failed.

- On Android, if you try to perform a write without response on a characteristic that doesn't support it, the `BluetoothGattCallback`'s `onCharacteristicWrite` method will be called with the `status` parameter indicating an error.

In either case, it's good practice to always check the properties of the characteristic before performing a write operation, to ensure that you're using a write operation that the characteristic supports. The properties can be checked using the `.properties` field in iOS or the `.getProperties()` method in Android."

are you sure your writes were actually working?

in 1.5.2 we ignored all errors

18723309899, 2023-07-29:
It's strange that when my device was previously controlled with Android, 'withoutResponses' was set to true to quickly respond to incoming' false 'responses. The response speed would be slow, and the IOS is different. I have always been sending in' false 'and can also respond quickly. It may be that our device has a bug but cannot be modified. We can only solve this problem on the Android side. Do you have a better solution

18723309899, 2023-07-29:
The version I used before was Flutter_ Blue_ Plus: ^ 1.3.1

chipweinberger, 2023-07-29:
You set false everywhere?

Try removing the checks and see what happens. 

##Debugging

The easiest way to debug issues in FlutterBluePlus is to make your own local copy.

`cd /user/downloads
git clone https://github.com/boskokg/flutter_blue_plus.git`
then in pubspec.yaml add the repo by path:

  `flutter_blue_plus:
    path: /user/downloads/flutter_blue_plus`
Now you can edit the FlutterBluePlus code yourself.

18723309899, 2023-07-29:
Okay, thank you very much

closed 2023-07-29 by 18723309899

automationsky, 2023-07-29:
After removing withoutResponse parameter. it worked fine most of the time. but it fails sometimes not sure why. like it fails after 8-10 write operation.

chipweinberger, 2023-07-29:
are you awaiting? make sure you do 'await chr.write(...)'

lmk if that is your problem^^