chipweinberger/flutter_blue_plus #449: [Bug]: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null)

18723309899, 2023-07-28:
### FlutterBluePlus Version

flutter_blue_plus: ^1.12.7

### What OS?

Android

### OS Version

Android 33

### What happened?

When I was using 'write' to send data, 'withResponses' passed in' ture 'and reported an error. I know it may be because my device does not support this feature, but I still want to enable it because I can set it to' ture 'in' flutter_blue_plus: ^ 1.5.2 '. This way, my device can send data quickly and I don't need to worry about packet loss. The problem now is that I need to wait for the device to respond for a long time, So I hope you can change back to the original state and allow me to set it freely. Thank you!

### Logs

```shell
I/flutter ( 5278): PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null)
I/flutter ( 5278): withoutResponse true
```

automationsky, 2023-07-28:
Sometimes face same issue with write operation. did you find any solution ?

chipweinberger, 2023-07-28:
duplicate of #448

closed 2023-07-28 by chipweinberger