chipweinberger/flutter_blue_plus #450: [Bug]:  Bluetooth autoconnect isue

Narayanhegde14, 2023-07-28:
### FlutterBluePlus Version

flutter_blue_plus 1.12.7

### What OS?

iOS

### OS Version

16.5.1(c)

### What happened?

After connection Write, Read everything is working, fine once the device is turned off and turned on Bluetooth reconnect is not happening.

Bluetooth Connect code 
```
await results[i].device.connect(autoConnect: true,);
```

### Logs

```shell
PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
I/flutter (12117): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1056)
I/flutter (12117): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:551)
I/flutter (12117): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
I/flutter (12117): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (12117): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (12117): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (12117): 	at android.os.Handler.handleCallback(Handler.java:942)
I/flutter (12117): 	at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (12117): 	at android.os.Looper.loopOnce(Looper.java:211)
I/flutter (12117): 	at android.os.Looper.loop(Looper.java:300)
I/flutter (12117): 	at
D/[FBP-Android](12117): [FBP-Android] onMethodCall: writeCharacteristic
```

chipweinberger, 2023-07-28:
auto connect requires ios 17

Narayanhegde14, 2023-07-28:
> auto connect requires ios 17
> 
> your logs are from android tho
> 
> make sure you are connected before you write
> 
> you can be disconnected even with auto connect

@chipweinberger  Yes in the latest version of the package in Android it's not reconnecting automatically. The log mentioned above is from this issue.

chipweinberger, 2023-07-28:
are you waiting for reconnection?

```
// wait for re-connection
`await device.connectionState.where((v) v == ConnectionState.connected).first;`

try: {
   chr.write(....)
} catch (e) {
   print(e); // maybe we lost connection during write
}
```

Narayanhegde14, 2023-07-29:
In the latest version of flutter_blue_plus for Android also auto-reconnect is not working. Is it an issue from package side or I missed something?

chipweinberger, 2023-07-29:
i'm not sure.

try turning on verbose logs.

chipweinberger, 2023-08-01:
did you fix it?

chipweinberger, 2023-08-02:
closing. please reopen if you still have issues.

closed 2023-08-02 by chipweinberger

Narayanhegde14, 2023-08-02:
> did you fix it?

No. I'm doing it manually by restarting the scan after disconnecting and doing await results[i].device.connect(autoConnect: false,);

chipweinberger, 2023-08-07:
>   ///   [autoConnect] Android only. reconnect whenever the device is found. This only
  ///   works if the device is in the Bluetooth cache or it is has been bonded before.

Narayanhegde14, 2023-08-19:
> > ///   [autoConnect] Android only. reconnect whenever the device is found. This only
> > ///   works if the device is in the Bluetooth scan cache or it is has been bonded before.
> > ///   The scan cache is cleared whenever bluetooth is turned off.

@chipweinberger 
Is there any way to clear the Bluetooth cache from the app? because when I did device.diconnect(); the Disconnected device was not showing in the scan result. I have to turn off and on my mobile Bluetooth to reconnect again. Same thing when I'm trying autoconnect: true while connecting. It's not reconnecting automatically..

chipweinberger, 2023-08-19:
>  the Disconnected device was not showing in the scan result

this is probably because your device stopped advertising.

Can any BLE scanner app can find your device?

Narayanhegde14, 2023-08-19:
> > the Disconnected device was not showing in the scan result
> 
> Can any BLE scanner app (even on a different phone) find your device? If not, your device is not advertising.

When i turn off my mobile Bluetooth then only it will be visible in other apps

chipweinberger, 2023-08-19:
> When i turn off my mobile Bluetooth then only it will be visible in other apps

Your device is not advertising, and is probably still connected to the first phone until you turn off bluetooth.

In addition to scanning, try looking through `FlutterBluePlus.connectedSystemDevices` . Maybe it is already connected?

Narayanhegde14, 2023-08-20:
> > When i turn off my mobile Bluetooth then only it will be visible in other apps
> 
> Your device is not advertising, and is probably still connected to the first phone until you turn off bluetooth.
> 
> In addition to scanning, try looking through `FlutterBluePlus.connectedSystemDevices` . Maybe it is already connected?
> 
> **Look at the README common problems section.**

Ok got it.. I have small question when i do autoconnect to true its not reconnecting and its not showing in the scan results in another app also.. any solution.. this is in new version of the package.. and in my Android 14 device

chipweinberger, 2023-08-20:
i've never used autoconnect, i am not sure