chipweinberger/flutter_blue_plus #451: [Bug]:

adsazad, 2023-07-28:
### FlutterBluePlus Version

1.9.3

### What OS?

Android

### OS Version

Android  13

### What happened?

I am trying to use `requestConnectionPriority`, I also tried using `setPreferredPhy` but it return the same sort of error.

```
    device!.requestConnectionPriority(connectionPriorityRequest: ConnectionPriority.high);
```
and 
```
device!.setPreferredPhy(txPhy: 1, rxPhy: 2,option: PhyOption.s8);
```

Complete function
```

  connectToDevice(deviceData) async {
    device = deviceData["device"];
    // await device!.disconnect();
    print("connecting");
    await device!.connect();
    print("connected");
    await device!.setPreferredPhy(txPhy: 1, rxPhy: 2, option: PhyOption.s8);

    await device!.requestConnectionPriority(
        connectionPriorityRequest: ConnectionPriority.high);

```

### Logs

```shell
Unhandled Exception: PlatformException(androidException, java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference, java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference
```

Same Exception on both functions.

E/flutter ( 7846): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference, java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference
E/flutter ( 7846): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:992)
E/flutter ( 7846): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/flutter ( 7846): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter ( 7846): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter ( 7846): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter ( 7846): 	at android.os.Handler.handleCallback(Handler.java:942)
E/flutter ( 7846): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter ( 7846): 	at android.os.Looper.loopOnce(Looper.java:226)
E/flutter ( 7846): 	at android.os.Looper.loop(Looper.java:313)
E/flutter ( 7846): 	at android.app.ActivityThread.main(ActivityThread.java:8757)
E/flutter ( 7846): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 7846): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/flutter ( 7846): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
E/flutter ( 7846): , null)
E/flutter ( 7846): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter ( 7846): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter ( 7846): <asynchronous suspension>
E/flutter ( 7846): #2      BluetoothDevice.requestConnectionPriority (package:flutter_blue_plus/src/bluetooth_device.dart:250:5)
E/flutter ( 7846): <asynchronous suspension>
```

adsazad, 2023-07-28:
Upgraded version to 1.12.7. but didn't work.

chipweinberger, 2023-07-28:
thanks for reporting i'll fix.

the problem is in bluetooth_msgs.dart

connectionPriority -> connection_priority

txPhy -> tx_phy
rxPhy -> rx_phy

etc

chipweinberger, 2023-07-28:
fixed in 1.12.8

closed 2023-07-28 by chipweinberger

adsazad, 2023-07-28:
thank you very much.
would this work on IOS?

chipweinberger, 2023-07-28:
no