chipweinberger/flutter_blue_plus #452: [Bug]: isOn not returning correct value on iOS 16.6 (v1.12.8)

keskink, 2023-07-28:
### FlutterBluePlus Version

1.12.8

### Flutter Version

3.10.xx

### What OS?

iOS

### OS Version

16.6

### What happened?

isOn function not returning correct value on iOS 16.6. 
works on 15.7.8.

### Logs

```shell
..
```

chipweinberger, 2023-07-28:
```
else if ([@"isOn" isEqualToString:call.method])
        {
            if (self.centralManager.state == CBManagerStatePoweredOn)
            {
                result(@(YES));
            }
            else
            {
                result(@(NO));
            }
        }
```

this code looks correct to me.

try turning on verbose logs

chipweinberger, 2023-07-28:
`FlutterBluePlus.adapterState` might be better to use

chipweinberger, 2023-07-28:
you can also try using `adapterState`

keskink, 2023-07-29:
FlutterBluePlus.adapterState.listen(
returning BluetoothAdapterState.unauthorized.
I am sure that all permission active.

keskink, 2023-07-29:
Logs as below

[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m<getConnectedSystemDevices>\^[[0m result: \^[[1;33m{devices: [<…>
flutter: [FBP] \^[[1;30m<getConnectedSystemDevices>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getConnectedSystemDevices>\^[[0m result: \^[[1;33m{devices: [<…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>

keskink, 2023-07-29:
I removed my app from phone, and restart app. it asked permissions again, then it is working now :)

closed 2023-07-31 by chipweinberger