chipweinberger/flutter_blue_plus #454: missing file AndroidManifest.xml

ICP-Theo, 2023-07-31:
### FlutterBluePlus Version

1.12.8

### Flutter Version

 3.10.6

### What OS?

Android

### OS Version

13

### What happened?

missing file AndroidManifest.xml in \android\src\main\

### Logs

```shell
* What went wrong:
A problem was found with the configuration of task ':flutter_blue_plus:processDebugManifest' (type 'ProcessLibraryManifest').
  - In plugin 'com.android.internal.version-check' type 'com.android.build.gradle.tasks.ProcessLibraryManifest' property 'mainManifest' specifies file '\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_blue_plus-1.12.8\android\src\main\AndroidManifest.xml' which doesn't exist.

    Reason: An input file was expected to be present but it doesn't exist.
```

dhruvanbhalara, 2023-07-31:
I've recently updated to the latest version, I'm not facing the issue. Can you please share more details to reproduce the issue?

MrCsabaToth, 2023-07-31:
This is weird. I'm using FBP 1.12.8, and my app builds, but now I'm looking at the main branch and it only has one java file in the src: https://github.com/boskokg/flutter_blue_plus/tree/master/android/src/main/java/com/boskokg/flutter_blue_plus

A few days ago there was a conversation where I noticed in a commit that the permissions are unnecessarily were duplicated from the main manifest into the debug and the profile manifest: https://github.com/boskokg/flutter_blue_plus/commit/c129565380d6226097b04884033e6f29d4e87e7a

In the conversation Chip asked if the debug and profile manifests can be removed: "both ChatGPT and Bard tells me that the debug and profile manifests can be removed. Since the default scaffolding contains the semi empty stubs maybe I'd just strip them down. It's up to you."

The duplications were removed 5 days ago: https://github.com/boskokg/flutter_blue_plus/commit/06c467d380a0f0590eef9ce917c436be089b2fab

Then later I think permissions were removed all together from the main manifest leaving the permissions settings all together to the user to try to satisfy colliding Android OS and app requirements:
* https://github.com/boskokg/flutter_blue_plus/issues/438#issuecomment-1652459928
* https://github.com/boskokg/flutter_blue_plus/issues/440#issuecomment-1652859361

But the complete disappearance of the manifest (I mean even the main manifest) doesn't sound right to me. Even though AGP 8.0+ now moves the namespace into the build.gradle, it's kinda weird if there's no manifest at all. Maybe this causes problem only with AGP < v8.0 folks?

chipweinberger, 2023-07-31:
@MrCsabaToth those commits are for the example app

for the plugin itself I completely removed the manifest, so people could easily set their own permissions.

We should figure out why some people can build fine.

MrCsabaToth, 2023-07-31:
I'm surprised that build succeeds without a main manifest. Even though app namespace is not in it any more because of AGP 8.0 moved it to build.gradle, I thought the Plugin java file needs to be referenced, but it looks like not: https://github.com/boskokg/flutter_blue_plus/blob/1.11.8/android/src/main/AndroidManifest.xml

chipweinberger, 2023-07-31:
should we add an empty main manifest?

MrCsabaToth, 2023-07-31:
Not sure.

chipweinberger, 2023-07-31:
@ICP-Theo @mvo5 

try running` flutter clean` & `flutter pub cache repair`

Jbz797, 2023-08-01:
Same here (on Flutter v2).

`flutter clean & flutter pub cache repair`don't work.

chipweinberger, 2023-08-01:
fixed 4ffdfac6106519807f9bb1f722fb67e8ea51306e

try master branch

```
dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/boskokg/flutter_blue_plus.git
      ref: master
```

chipweinberger, 2023-08-01:
should be fixed in 1.12.9

reopen if you still have issues.

closed 2023-08-01 by chipweinberger