chipweinberger/flutter_blue_plus #456: [Bug]:

amasuke55, 2023-08-01:
### FlutterBluePlus Version

1.12.8

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

13

### What happened?

I started the example code and connected to a device. The services were discovered and shown.
I went back to the find devices view and opened the connected device via the open button, the services were still shown.
When I disconnected the device and then connected it again in the find devices view, I could not get the services to appear again.

Discorver Services also sometimes gives 

I expected the services to always be discoverable and shown when hitting the corresponding button, as it was in version 1.8.8.

### Logs

```shell
D/[FBP-Android](18301): [FBP-Android] onMethodCall: connect
I/BluetoothAdapter(18301): STATE_ON
D/BluetoothGatt(18301): connect() - device: A4C69A_6, auto: false
D/BluetoothGatt(18301): registerApp()
D/BluetoothGatt(18301): registerApp() - UUID=85cf86fb-b381-42c6-b6d1-7ae73b37aa47
D/BluetoothGatt(18301): onClientRegistered() - status=0 clientIf=7
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/BluetoothGatt(18301): onClientConnectionState() - status=0 clientIf=6 device=A4C69A_6
D/BluetoothGatt(18301): onClientConnectionState() - status=0 clientIf=7 device=A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onConnectionStateChange: status: 0 newState: connected
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android](18301): [FBP-Android] onConnectionStateChange: status: 0 newState: connected
D/[FBP-Android](18301): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18301): readRssi() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onReadRemoteRssi: rssi: -55 status: 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(18301): discoverServices() - device: A4C69A_6
D/BluetoothGatt(18301): onSearchComplete() = Device=A4C69A_6 Status=0
D/[FBP-Android](18301): [FBP-Android] onServicesDiscovered: count: 2 status: 0
D/BluetoothGatt(18301): onPhyUpdate() - status=0 address=A4C69A_6 txPhy=2 rxPhy=2
D/BluetoothGatt(18301): onPhyUpdate() - status=0 address=A4C69A_6 txPhy=2 rxPhy=2
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18301): readRssi() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onReadRemoteRssi: rssi: -58 status: 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android](18301): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18301): readRssi() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onReadRemoteRssi: rssi: -56 status: 0
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(18301): discoverServices() - device: A4C69A_6
D/BluetoothGatt(18301): onSearchComplete() = Device=A4C69A_6 Status=0
D/[FBP-Android](18301): [FBP-Android] onServicesDiscovered: count: 2 status: 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: disconnect
D/BluetoothGatt(18301): cancelOpen() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/BluetoothGatt(18301): onClientConnectionState() - status=0 clientIf=7 device=A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onConnectionStateChange: status: 0 newState: disconnected
D/BluetoothGatt(18301): close()
D/BluetoothGatt(18301): unregisterApp() - mClientIf=7
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: connect
I/BluetoothAdapter(18301): STATE_ON
D/BluetoothGatt(18301): connect() - device: A4C69A_6, auto: false
D/BluetoothGatt(18301): registerApp()
D/BluetoothGatt(18301): registerApp() - UUID=fbc62112-6e45-4ddb-a419-bfdf0e75b95f
D/BluetoothGatt(18301): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(18301): onClientConnectionState() - status=0 clientIf=7 device=A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onConnectionStateChange: status: 0 newState: connected
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android](18301): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18301): readRssi() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onReadRemoteRssi: rssi: -60 status: 0
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(18301): discoverServices() - device: A4C69A_6
D/BluetoothGatt(18301): onSearchComplete() = Device=A4C69A_6 Status=0
D/[FBP-Android](18301): [FBP-Android] onServicesDiscovered: count: 2 status: 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(18301): discoverServices() - device: A4C69A_6
D/BluetoothGatt(18301): onSearchComplete() = Device=A4C69A_6 Status=0
D/[FBP-Android](18301): [FBP-Android] onServicesDiscovered: count: 2 status: 0
D/[FBP-Android](18301): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18301): readRssi() - device: A4C69A_6
D/[FBP-Android](18301): [FBP-Android] onReadRemoteRssi: rssi: -58 status: 0
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 0
I/ViewRootImpl@2df67b[MainActivity](18301): ViewPostIme pointer 1
D/[FBP-Android](18301): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](18301): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(18301): discoverServices() - device: A4C69A_6
D/BluetoothGatt(18301): onSearchComplete() = Device=A4C69A_6 Status=0
D/[FBP-Android](18301): [FBP-Android] onServicesDiscovered: count: 2 status: 0
```

chipweinberger, 2023-08-01:
Seems like a UI issue.

Can you fix it and open a PR?

chipweinberger, 2023-08-01:
try 1.12.9

chipweinberger, 2023-08-07:
This may be due to us not considering Bond state after a connection

https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07

[image]

chipweinberger, 2023-08-08:
I've release `1.13.0` which handles bonded devices, and I think was the cause of this issue.

Reopen a _new_ issue if you still have problems.

closed 2023-08-08 by chipweinberger

amasuke55, 2023-08-09:
> I've release `1.13.0` which handles bonded devices, and I think was the cause of this issue.
> 
> Reopen a new issue if you still have problems.

Hello, sorry for the late response. I was looking into what could have caused the issue and then tried the newer version, which fixed the issue but introduced a new one, where the services got stuck. So disconnecting and then connecting again, the services were automatically there, which was odd, they also didn't work I still had to discover them again for them to work. Which is still a thing, but only when the user stays on the Device Screen, which doesn't matter to me. Before it was also when going back to the Find Devices screen.
I tested Version 1.14.0 now and can not reproduce the original issue so it is indeed fixed.

chipweinberger, 2023-08-09:
I do not recommend 1.13.x

1.14.0 should be better