chipweinberger/flutter_blue_plus #459: [Question]: How to listen `BluetoothDevice.connectionState` if connect device twice?

leimenghao1991, 2023-08-01:
### Ask your question

`BluetoothDevice.connectionState` is a stream, I listen it when user open device page. I Call `BluetoothDevice.disconnect` when user exist device page. But when user re-enter this device page, `BluetoothDevice.connectState.listen` is called and throw an error: "Bad state: Stream has already been listened to". 
So in this situation how can i observe connect state every time user enter device page?

chipweinberger, 2023-08-01:
`connectionState` is a single subscription stream. It can only be listened to _once_ ever.

I'm going to push a fix for this.

chipweinberger, 2023-08-01:
I'm not sure what is causing that. `device.connectionState` returns a new single subscription stream every time it is called.

leimenghao1991, 2023-08-02:
Sorry for misdescription. In fact, `device.servicesStream` throw the error message. Logs are below:
```
E/flutter ( 6530): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Bad state: Stream has already been listened to.
E/flutter ( 6530): #0      _StreamController._subscribe (dart:async/stream_controller.dart:678:7)
E/flutter ( 6530): #1      _ControllerStream._createSubscription (dart:async/stream_controller.dart:828:19)
E/flutter ( 6530): #2      _StreamImpl.listen (dart:async/stream_impl.dart:471:9)
E/flutter ( 6530): #3      new _AddStreamState (dart:async/stream_controller.dart:893:34)
E/flutter ( 6530): #4      new _StreamControllerAddStreamState (dart:async/stream_controller.dart:943:9)
E/flutter ( 6530): #5      _StreamController.addStream (dart:async/stream_controller.dart:578:9)
E/flutter ( 6530): #6      _AsyncStarStreamController.addStream (dart:async-patch/async_patch.dart:98:39)
E/flutter ( 6530): #7      BluetoothDevice.servicesStream (package:flutter_blue_plus/src/bluetooth_device.dart)
E/flutter ( 6530): <asynchronous suspension>
E/flutter ( 6530): #8      new BleDevice.<anonymous closure> (package:bluetooth_connect/device/bledevice.dart:21:56)
E/flutter ( 6530): <asynchronous suspension>
E/flutter ( 6530): 
```

My BleDevice code: 
```
class BleDevice {
  final BluetoothDevice device;
  late StreamSubscription<BluetoothConnectionState> stateSubscription;
  final BehaviorSubject<BluetoothConnectionState> _stateSubject =
      BehaviorSubject.seeded(BluetoothConnectionState.disconnected);
  final BehaviorSubject<List<BluetoothService>> _serviceSubject = BehaviorSubject();
  late StreamSubscription<List<BluetoothService>> serviceSubscription;
  bool isDiscovering = false;

  BleDevice(this.device) {
    stateSubscription = device.connectionState.listen((event) {
      _stateSubject.add(event);
    });
    serviceSubscription = device.servicesStream.listen((event) {
      _serviceSubject.add(event);
    });
  }

  List<BluetoothService> services = [];

  Future<bool> connect() async {
    try {
      _stateSubject.add(BluetoothConnectionState.connecting);
      await device.connect(
          timeout: const Duration(seconds: 20), autoConnect: false);
    } catch (e) {
      debugPrint("error connect: $e");
    }

    try {
      debugPrint("connect result: $state");
      bool connectResult = state == BluetoothConnectionState.connected;
      _stateSubject.add(BluetoothConnectionState.connected);
      return connectResult;
    } catch (e) {
      debugPrint("check state failed: $e");
    }
    return false;
  }

  Future<bool> disconnect() async {
    _stateSubject.add(BluetoothConnectionState.disconnecting);
    try {
      await device.disconnect();
    } catch (e) {
      debugPrint("disconnect failed: $e");
    }
    _stateSubject.add(BluetoothConnectionState.disconnected);
    debugPrint("disconnect finish");
    return true;
  }

  bool isConnected() => state == BluetoothConnectionState.connected;

  bool isConnecting() => state == BluetoothConnectionState.connecting;

  Future<bool> discoverService() async {
    if (isDiscovering) {
      debugPrint("already in discovering!!");
      return true;
    }
    isDiscovering = true;
    try {
      services = await device.discoverServices();
    } catch(e) {
      debugPrint("discover service failed: $e");
      _serviceSubject.add([]);
    }
    isDiscovering = false;
    return services.isNotEmpty;
  }

  BluetoothConnectionState get state => _stateSubject.value;
  
  Stream<BluetoothConnectionState> get stateStream => _stateSubject.distinct();

  Stream<List<BluetoothService>> get serviceStream => _serviceSubject;

  void destroy() {
    stateSubscription.cancel();
    serviceSubscription.cancel();
    debugPrint("${device.addressOrDefault()} destroy");
  }
}

```

leimenghao1991, 2023-08-02:
I changed constructor method to:
```
BleDevice(this.device) {
    stateSubscription = device.connectionState.listen((event) {
      _stateSubject.add(event);
    });
    var currentService = device.servicesList;
    debugPrint("services: $currentService");
    if (currentService != null) {
      _serviceSubject.add(currentService);
      isDiscovering = true;
    } else {
      serviceSubscription = device.servicesStream.listen((event) {
        _serviceSubject.add(event);
      });
    }
  }
```
And log print:
```
I/flutter ( 8076): services: null
E/flutter ( 8076): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Bad state: Stream has already been listened to.
E/flutter ( 8076): #0      _StreamController._subscribe (dart:async/stream_controller.dart:678:7)
E/flutter ( 8076): #1      _ControllerStream._createSubscription (dart:async/stream_controller.dart:828:19)
E/flutter ( 8076): #2      _StreamImpl.listen (dart:async/stream_impl.dart:471:9)
E/flutter ( 8076): #3      new _AddStreamState (dart:async/stream_controller.dart:893:34)
E/flutter ( 8076): #4      new _StreamControllerAddStreamState (dart:async/stream_controller.dart:943:9)
E/flutter ( 8076): #5      _StreamController.addStream (dart:async/stream_controller.dart:578:9)
E/flutter ( 8076): #6      _AsyncStarStreamController.addStream (dart:async-patch/async_patch.dart:98:39)
E/flutter ( 8076): #7      BluetoothDevice.servicesStream (package:flutter_blue_plus/src/bluetooth_device.dart)
E/flutter ( 8076): <asynchronous suspension>
E/flutter ( 8076): #8      new BleDevice.<anonymous closure> (package:bluetooth_connect/device/bledevice.dart:27:58)
E/flutter ( 8076): <asynchronous suspension>
E/flutter ( 8076): 
```
It seems that, the device is a new instance and it's service list is null, but the `servicesStream` is not new instance. I checked the source code, it's indeed a cache value.

leimenghao1991, 2023-08-02:
So must rescan to generate a new device when user exist device page and reselect this device?

chipweinberger, 2023-08-02:
`servicesStream` is fixed in 1.12.9

😀

closed 2023-08-02 by chipweinberger