chipweinberger/flutter_blue_plus #462: [Bug]: After pair() removeBond(), app crash

juggCan, 2023-08-02:
### FlutterBluePlus Version

1.12.9

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

11

### What happened?

1.After pair() removeBond(), app crash
2.When connecting to a bond, turning on or off Bluetooth devices will cause the app to crash

### Logs

```shell
2023-08-02 23:41:25.830 9387-9387/xxx D/[FBP-Android]: onAttachedToEngine
2023-08-02 23:41:25.831 9387-9387/xxx D/[FBP-Android]: onAttachedToActivity
2023-08-02 23:41:26.963 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: setLogLevel
2023-08-02 23:41:26.964 9387-9387/xxx D/[FBP-Android]: [FBP-Android] initializing BluetoothAdapter
2023-08-02 23:41:26.964 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: isAvailable
2023-08-02 23:41:27.842 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: turnOn
2023-08-02 23:41:27.844 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getAdapterState
2023-08-02 23:41:27.907 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getAdapterState
2023-08-02 23:41:27.925 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectedSystemDevices
2023-08-02 23:41:28.285 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getBondedDevices
2023-08-02 23:41:34.752 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: startScan
2023-08-02 23:41:47.515 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: connect
2023-08-02 23:41:47.535 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:41:47.536 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:41:48.155 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onConnectionStateChange: status: 0 newState: connected
2023-08-02 23:41:48.203 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:41:48.207 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: discoverServices
2023-08-02 23:41:48.219 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onServicesDiscovered: count: 7 status: 0
2023-08-02 23:41:48.296 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: setNotification
2023-08-02 23:41:48.350 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
2023-08-02 23:41:48.371 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: writeCharacteristic
2023-08-02 23:41:48.446 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onCharacteristicChanged: uuid: 6e40fa32-b5a3-f393-e0a9-e50e24dcca9e
2023-08-02 23:41:48.449 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onCharacteristicWrite: uuid: 6e40fa31-b5a3-f393-e0a9-e50e24dcca9e status: 0
2023-08-02 23:41:53.527 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: createBond
2023-08-02 23:41:59.874 9387-9387/xxx D/[FBP-Android]: onDetachedFromActivityForConfigChanges
2023-08-02 23:41:59.874 9387-9387/xxx D/[FBP-Android]: onDetachedFromActivity
2023-08-02 23:41:59.875 9387-9387/xxx D/[FBP-Android]: onDetachedFromEngine
2023-08-02 23:41:59.876 9387-9387/xxx D/[FBP-Android]: calling disconnect() on device: D7:35:26:8E:41:B4
2023-08-02 23:41:59.876 9387-9387/xxx D/[FBP-Android]: calling gatt.close() on device: D7:35:26:8E:41:B4
2023-08-02 23:42:00.441 9387-9387/xxx D/[FBP-Android]: onAttachedToEngine
2023-08-02 23:42:00.443 9387-9387/xxx D/[FBP-Android]: onAttachedToActivity
2023-08-02 23:42:01.597 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: setLogLevel
2023-08-02 23:42:01.597 9387-9387/xxx D/[FBP-Android]: [FBP-Android] initializing BluetoothAdapter
2023-08-02 23:42:01.599 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: isAvailable
2023-08-02 23:42:02.543 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: turnOn
2023-08-02 23:42:02.545 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getAdapterState
2023-08-02 23:42:02.630 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getAdapterState
2023-08-02 23:42:02.650 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectedSystemDevices
2023-08-02 23:42:02.763 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getBondedDevices
2023-08-02 23:42:03.042 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: connect
2023-08-02 23:42:03.047 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:42:03.048 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:42:03.057 9387-9409/xxx D/[FBP-Android]: [FBP-Android] onConnectionStateChange: status: 0 newState: connected
2023-08-02 23:42:03.150 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-02 23:42:03.153 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: discoverServices
2023-08-02 23:42:03.160 9387-9409/xxx D/[FBP-Android]: [FBP-Android] onServicesDiscovered: count: 7 status: 0
2023-08-02 23:42:03.217 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: setNotification
2023-08-02 23:42:04.078 9387-9409/xxx D/[FBP-Android]: [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
2023-08-02 23:42:04.101 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: writeCharacteristic
2023-08-02 23:42:05.118 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onCharacteristicChanged: uuid: 6e40fa32-b5a3-f393-e0a9-e50e24dcca9e
2023-08-02 23:42:05.125 9387-9408/xxx D/[FBP-Android]: [FBP-Android] onCharacteristicWrite: uuid: 6e40fa31-b5a3-f393-e0a9-e50e24dcca9e status: 0
```

chipweinberger, 2023-08-02:
where is the crash?

there is no crash in the logs.

juggCan, 2023-08-03:
Focus on the following information，After pair() removeBond()-》 onDetachedFromActivityForConfigChanges -》onDetachedFromActivity
2023-08-02 23:41:53.527 9387-9387/xxx D/[FBP-Android]: [FBP-Android] onMethodCall: createBond
2023-08-02 23:41:59.874 9387-9387/xxx D/[FBP-Android]: onDetachedFromActivityForConfigChanges

There is no clear log，
Only:
2023-08-03 10:01:12.370 28321-28321/XXX I/BluetoothDevice: removeBond() for device D7:35:26:8E:41:B4 called by pid: 28321 tid: 28321
2023-08-03 10:01:12.391 28321-28321/XXX D/[FBP-Android]: [FBP-Android] onMethodCall: getConnectionState
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: │ #0   JcLogger.d (package:pentool/utils/jc_logger.dart:8:12)
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: │ #1   new DeviceBeanCp.<anonymous closure> (package:pentool/models/device_bean.dart:19:16)
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: │ 🐛  device.state :BluetoothConnectionState.disconnected
2023-08-03 10:01:12.393 28321-28361/XXX I/flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.427 28321-28361/XXX I/flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.427 28321-28361/XXX I/flutter: │ #0   JcLogger.d (package:pentool/utils/jc_logger.dart:8:12)
2023-08-03 10:01:12.427 28321-28361/XXX I/flutter: │ #1   DeviceHome.build.<anonymous closure>.<anonymous closure> (package:pentool/pages/device_home.dart:33:22)
2023-08-03 10:01:12.427 28321-28361/XXX I/flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
2023-08-03 10:01:12.428 28321-28361/XXX I/flutter: │ 🐛 itemBuilder!!!!!!~~~~~0
2023-08-03 10:01:12.428 28321-28361/XXX I/flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: │ #0   JcLogger.d (package:pentool/utils/jc_logger.dart:8:12)
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: │ #1   DeviceHome.build.<anonymous closure>.<anonymous closure> (package:pentool/pages/device_home.dart:33:22)
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: │ 🐛 itemBuilder!!!!!!~~~~~1
2023-08-03 10:01:12.517 28321-28361/XXX I/flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-08-03 10:01:12.744 28321-28321/XXX D/[FBP-Android]: onDetachedFromActivityForConfigChanges
2023-08-03 10:01:12.744 28321-28321/XXX D/[FBP-Android]: onDetachedFromActivity
2023-08-03 10:01:12.746 28321-28321/XXX D/[FBP-Android]: onDetachedFromEngine
2023-08-03 10:01:13.273 28321-28363/XXX E/libEGL: call to OpenGL ES API with no current context (logged once per thread)
2023-08-03 10:01:13.335 28321-28321/XXX D/[FBP-Android]: onAttachedToEngine
2023-08-03 10:01:13.337 28321-28321/XXX D/[FBP-Android]: onAttachedToActivity
2023-08-03 10:01:13.422 28321-28321/XXX I/Choreographer: Skipped 64 frames!  The application may be doing too much work on its main thread.
2023-08-03 10:01:13.471 28321-28321/XXX E/SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length

chipweinberger, 2023-08-03:
So it is not crashing? 

It is causing the application to re-launch the activity?

I am not sure what would cause that. You may need to debug this yourself.

## Debugging

The easiest way to debug issues in FlutterBluePlus is to make your own local copy.

```
cd /user/downloads
git clone https://github.com/boskokg/flutter_blue_plus.git
```

then in `pubspec.yaml` add the repo by path:

```
  flutter_blue_plus:
    path: /user/downloads/flutter_blue_plus
```

Now you can edit the FlutterBluePlus code yourself.

chipweinberger, 2023-08-03:
The relaunch may be expected behavior. When Android shows system dialogue boxes it may start a new Activity, causing your Activity to end. 

https://developer.android.com/training/basics/intents/result

juggCan, 2023-08-04:
After reviewing a lot of information and analyzing the current situation, I finally identified the problem as

android:configChanges

Flutter default：
android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"

need to add navigation
 android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode|navigation"

I tested the demo and it should be possible.

My device is an external keyboard, and as for why I added 'navigation', I didn't check the documentation anymore...

chipweinberger, 2023-08-04:
From ChatGPT:

> The navigation configuration change in Android is used to handle changes in the type of navigation method used in the device.
> 
> navigation: The navigation type (trackball/dpad) has changed.
> 
> This configuration change occurs when the user switches the input method on the device. For example, if a user switches from using a trackball to using a dpad (directional pad), this would trigger a navigation configuration change.
> 
> Remember, when a configuration change occurs, by default Android will destroy and recreate the activity with the new configuration. If you declare navigation in android:configChanges, it means that you will handle this configuration change yourself, and Android will not restart your activity. Instead, it will call the onConfigurationChanged() method of your activity.
> 
> As of my knowledge cutoff in September 2021, this is a pretty uncommon configuration change, as most Android devices don't readily allow users to switch navigation types. However, it could potentially be more relevant on devices like Android TVs or devices with physical keyboards and trackballs.

So by specifying `navigation`,  it means that you will handle this configuration change.  and Android will not restart your activity. Instead, it will call the `onConfigurationChanged()` method of your activity.

chipweinberger, 2023-08-04:
> external keyboard

Typically external keyboards do not support BLE. From the FlutterBluePlus readme:

## ❗ Bluetooth Classic is not supported ❗

 i.e. speakers, headphones, mice, keyboards, gamepads, Arduino HC-05 & HC-06, and more are not supported

chipweinberger, 2023-08-04:
glad you figured it out.

closed 2023-08-04 by chipweinberger