chipweinberger/flutter_blue_plus #464: [Question]: Android BT turned off and on can't reconnect

lwd369, 2023-08-03:
### Ask your question

I encountered  (maybe) the same [issue](https://github.com/boskokg/flutter_blue_plus/issues/214#issue-1606339733).

version 1.12.7 still meet this problem.

Even I've called clearGattCache() after connect.

```
D/BluetoothGatt(25078): discoverServices() - device: C4:64:E3:50:61:09
3
E/BluetoothGatt(25078): android.os.DeadObjectException
E/BluetoothGatt(25078): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt(25078): 	at android.os.BinderProxy.transact(BinderProxy.java:584)
E/BluetoothGatt(25078): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.discoverServices(IBluetoothGatt.java:1603)
E/BluetoothGatt(25078): 	at android.bluetooth.BluetoothGatt.discoverServices(BluetoothGatt.java:1124)
E/BluetoothGatt(25078): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:497)
E/BluetoothGatt(25078): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/BluetoothGatt(25078): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/BluetoothGatt(25078): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/BluetoothGatt(25078): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/BluetoothGatt(25078): 	at android.os.Handler.handleCallback(Handler.java:942)
E/BluetoothGatt(25078): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BluetoothGatt(25078): 	at android.os.Looper.loopOnce(Looper.java:201)
E/BluetoothGatt(25078): 	at android.os.Looper.loop(Looper.java:288)
E/BluetoothGatt(25078): 	at android.app.ActivityThread.main(ActivityThread.java:7918)
E/BluetoothGatt(25078): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt(25078): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/BluetoothGatt(25078): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

Connected -> turn off BT -> turn on BT -> rescan -> connect (FAIL)

I also tried add a sleep after BT is turned on , but still deadobject. 

The original flutter_blue package dose not have this problem.

Has any idea?

chipweinberger, 2023-08-03:
[image]

chipweinberger, 2023-08-03:
please share all logs

chipweinberger, 2023-08-03:
I've merged this fix e22e19983b5087f52f23e7b8d78654c0258d3ad4

please try master branch

dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/boskokg/flutter_blue_plus.git
      ref: master

chipweinberger, 2023-08-03:
release 1.12.11

reopen if it does not fix the problem

closed 2023-08-03 by chipweinberger

lwd369, 2023-08-04:
Thank you for fixing the bug! Your help is greatly appreciated.

chipweinberger, 2023-08-04:
no problem - did it fix it?

lwd369, 2023-08-04:
Yes, it's fixed.