chipweinberger/flutter_blue_plus #465: [Question]: Error on discoverServices

umuttsaltan, 2023-08-03:
### Ask your question

After i connect a device, i use this function to get power data from BLE device. 

Function : 

void powerChar() async {
    List<BluetoothDevice> connectedDevices =
        await FlutterBluePlus.connectedSystemDevices;
    BluetoothDevice powerDevice = connectedDevices.first;
    List<BluetoothService> powerServices = await powerDevice.discoverServices();

    for (var servis in powerServices) {
      if (servis.uuid.toString() == "0000181a-0000-1000-8000-00805f9b34fb") {
        for (var ch in servis.characteristics) {
          if (ch.uuid.toString() == "7e422156-ab78-11ed-afa1-0242ac120002") {
            powerCharacteristic = ch;
            Timer(const Duration(milliseconds: 500), () async {
              try {
                await powerCharacteristic.setNotifyValue(true);
              } catch (e) {
                print("Bir hata oluştu: $e");
              }
            });
          }
        }
      }
    }

    Timer(const Duration(milliseconds: 500), () async {
      powerCharacteristic.onValueReceived.listen((power) {
        if (power.length == 2) {
          int powerdata = (power[0] << 8) + power[1];
          _devicedata.values.elementAt(currentDeviceIndex!).devicePower =
              powerdata;
          
        }
      });
    });
  }

When i'm running the app on iphone, there's no problem with function but on android it throws error on discoverServices. Am i doing something wrong? 

Error Log :

E/flutter ( 5264): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
E/flutter ( 5264):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1050)
E/flutter ( 5264):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:489)
E/flutter ( 5264):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/flutter ( 5264):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter ( 5264):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter ( 5264):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter ( 5264):      at android.os.Handler.handleCallback(Handler.java:938)
E/flutter ( 5264):      at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter ( 5264):      at android.os.Looper.loop(Looper.java:223)
E/flutter ( 5264):      at android.app.ActivityThread.main(ActivityThread.java:8011)
E/flutter ( 5264):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 5264):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:603)
E/flutter ( 5264):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
E/flutter ( 5264): , null)
E/flutter ( 5264): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652)
E/flutter ( 5264): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310)
E/flutter ( 5264): <asynchronous suspension>
E/flutter ( 5264): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:279)
E/flutter ( 5264): <asynchronous suspension>
E/flutter ( 5264): #3      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:113)
E/flutter ( 5264): <asynchronous suspension>
E/flutter ( 5264): #4      BleFunctions.powerChar (package:makelhome2/controllers/ble_functions.dart:123)
E/flutter ( 5264): <asynchronous suspension>

chipweinberger, 2023-08-03:
you need to call:

```
    await powerDevice.connect();
    List<BluetoothService> powerServices = await powerDevice.discoverServices();
```

theyre connected to the system, but you still need to connect your app

closed 2023-08-03 by chipweinberger

umuttsaltan, 2023-08-04:
> you need to call:
> 
> ```
>     await powerDevice.connect();
>     List<BluetoothService> powerServices = await powerDevice.discoverServices();
> ```
> 
> theyre connected to the system, but you still need to connect your app

i tried this like that but still same problem