chipweinberger/flutter_blue_plus #466: [Question]: Issue regarding "FlutterBluePlus flutterBlue = FlutterBluePlus.instance;"

100burn, 2023-08-03:
### Ask your question

Thank you for the good package. 

I'm still using 1.1.3 ver.
suddenly a problem occured on Android Studio in iOS,

"This expression has a type of 'void' so its value can't be used. "

The line is
"FlutterBluePlus flutterBlue = FlutterBluePlus.instance;"

and another problem occured.
"The sataic method 'startScan' can't be accessed through an instance."

chipweinberger, 2023-08-03:
screenshots? please update

100burn, 2023-08-03:
FlutterBluePlus flutterBlue = FlutterBluePlus.instance;

  /* Scan */
  void scan() async {
    if (!isScanning) {  //if is not scanning, start scan
      productList.clear();
      addressList.clear();
      setState(() {});
      // Listen to scan results
      isScanning = true;
      flutterBlue.startScan(
          scanMode: ScanMode(2), allowDuplicates: true);
      flutterBlue.scanResults.listen((results) {
        // do something with scan results
        scanResultList = results;

        for (ScanResult s in results) {
          if (isProduct(s)) {
            // ToastText(CalcScale(s).toString());
            String name = s.device.name;
            String address = s.device.id.id;

            if(!addressList.contains(address)){
              ProductData pd = ProductData(name, address);

              productList.add(pd);
              addressList.add(address);
              setState(() {});
            }

            }
            else{

            }
          }
        });
      // update state
    }
    else{
      isScanning = false;
      flutterBlue.stopScan();
      setState(() {});
    }
  }

chipweinberger, 2023-08-03:
I think you updated and did not realize it.

You do not need to do `FlutterBluePlus.instance` anymore.

Just do `FlutterBluePlus.startScan`, `FlutterBluePlus.scanResults`, etc.

closed 2023-08-03 by chipweinberger

100burn, 2023-08-03:
Thank you!! Gorgeous!!

100burn, 2023-08-31:
Hello, Sir?

I have a question.

for using higher version of package,
How do I change the code?

I used "flutter_blue_plus: ^1.1.3" until now;
I had to upgrade the version of package(FlutterBluePlus)

I tried to use ^1.14.11
I changed  flutterBlue to  FlutterBluePlus.

But I couldn't get the any signal from my signal;

To get the signal, I used listen.
 FlutterBluePluse.scanResults.listen

Is it still working on 1.14.11?
I want to get the manufacturerData from device
using 'r.advertisementData.manufacturerData'

-------------------------------------------------------------
  void scan() async {
    if (!isScanning) {  
      productList.clear();
      addressList.clear();

      // Listen to scan results
      isScanning = true;
      FlutterBluePlus.startScan(
          scanMode: ScanMode(2), allowDuplicates: true);
      FlutterBluePluse.scanResults.listen((results) {

        for (ScanResult s in results) {
          if (isMydevice(s)) {

            String name = s.device.name;
            String address = s.device.id.id;

            if(!ApiLoginData.productList.containsValue(address)){
            if (!addressList.contains(address)) {

              ProductData pd = ProductData(name, address);

              productList.add(pd);
              addressList.add(address);
              setState(() {});
            }
          }
            }
            else{
            }
          }
        });
      // update state
    }
    else{
      isScanning = false;
      FlutterBluePlus.stopScan();
      setState(() {});
    }
  }

bool isMydevice(ScanResult r){

    try {
    if(r == null ||
        r.advertisementData == null ||
        r.advertisementData.manufacturerData == null){
      return false;
    }

    if ((r.advertisementData.manufacturerData.keys.first & 0xC0) != 0xC0)
    {
      return false;
    }

    if (r
        .advertisementData
        .manufacturerData[r.advertisementData.manufacturerData.keys.first]!
        .length ==
        13) {
      return true;
    } else {
      return false;
    }

    } catch (e) {}
    return false;
  }
-------------------------------------------------------------

ritish-metta, 2024-12-28:
import 'package:convert/convert.dart'; // Library for hex encoding/decoding.
import 'package:flutter/material.dart'; // Core library for building Flutter UI.
import 'package:flutter_blue_plus/flutter_blue_plus.dart'; // Bluetooth plugin for Flutter.
import 'package:modal_progress_hud_nsn/modal_progress_hud_nsn.dart'; // Plugin for showing a loading overlay.

// Main widget for the bulb control page.
class BulbPage extends StatefulWidget {
  const BulbPage({super.key});

  @override
  State<BulbPage> createState() => _BulbPageState();
}

class _BulbPageState extends State<BulbPage> {
  // UUIDs for Bluetooth service and characteristic specific to the bulb.
  final String serviceUuid = '0000ffe0-0000-1000-8000-00805f9b34fb';
  final String characteristicUuid = '0000ffe1-0000-1000-8000-00805f9b34fb';

  // State variables
  bool _isBulbOn = false; // Tracks bulb's on/off state.
  bool isloading = false; // Shows loading state during operations.
  BluetoothDevice? connectionDevice; // Stores the connected Bluetooth device.
  BluetoothCharacteristic? currentCharacteristic; // Stores the active characteristic for communication.

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      // Change background color based on bulb state.
      backgroundColor: _isBulbOn
          ? const Color.fromARGB(255, 214, 18, 18) // Red if bulb is on.
          : const Color.fromARGB(255, 253, 242, 242), // Light color if off.
      body: ModalProgressHUD(
        inAsyncCall: isloading, // Show loading overlay during async operations.
        child: Center(
          child: Column(
            mainAxisAlignment: MainAxisAlignment.center,
            children: [
              // Display the current state of the bulb.
              Text(
                _isBulbOn ? 'Bulb is ON' : 'Bulb is OFF',
                style: TextStyle(
                  color: _isBulbOn
                      ? const Color.fromARGB(255, 37, 210, 18) // Green text if on.
                      : const Color.fromARGB(255, 119, 13, 231), // Purple text if off.
                  fontSize: 24,
                ),
              ),
              const SizedBox(height: 20),
              // Circular button to toggle the bulb state.
              Container(
                width: 120,
                height: 120,
                decoration: BoxDecoration(
                  color: _isBulbOn
                      ? const Color.fromARGB(255, 37, 210, 18) // Green if on.
                      : const Color.fromARGB(255, 119, 13, 231), // Purple if off.
                  shape: BoxShape.circle,
                  boxShadow: [
                    BoxShadow(
                      color: _isBulbOn
                          ? const Color.fromARGB(255, 37, 210, 18)
                          : const Color.fromARGB(255, 119, 13, 231),
                      blurRadius: 10,
                      offset: const Offset(0, 4),
                    ),
                  ],
                ),
                child: IconButton(
                  // Icon changes based on bulb state.
                  icon: Icon(
[... 110 lines cut ...]
                            //print(value);
                            final actualvalue = int.parse(value, radix: 16);
                            //print(actualvalue);
                            _isBulbOn = actualvalue == 1;
                            connectionDevice = device;
                          }
                        }

                        setState(() {
                          isloading = false; // Stop loading spinner.
                        });
                      },
                child: Text(
                  connectionDevice != null ? 'Disconnect' : 'Connect',
                  style: const TextStyle(color: Colors.white),
                ),
              ),
            ],
          ),
        ),
      ),
    );
  }
}
(iam getting even removed it the instance)