chipweinberger/flutter_blue_plus #467: [Question]: Cannot find Bluetooth devices using Android 13 and latest version of plugin (1.12.11)

Indigi-managed, 2023-08-04:
### Ask your question

Hoping someone can assist, I can no longer discover Bluetooth devices uses this package in RELEASE mode, DEBUG mode works fine for some reason but when I can for Bluetooth devices using the latest package in release mode, nothing happens.

I'm testing on Android 13 on a Samsung Galaxy S22+, here is my AndroidManifest:

     <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

     
     <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
     <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

 
     
     <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
     <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
     <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="33"/>
 
 
     
     <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="33" />
     
     I'm not an expert at Android development so please let me know if I'm missing anything.

chipweinberger, 2023-08-04:
this is a bug report. 

please use the bug report form.

chipweinberger, 2023-08-04:
You can try this

### Android Proguard

Add the following line in your `project/android/app/proguard-rules.pro` file:

```
-keep class com.boskokg.flutter_blue_plus.* { *; }
```

to avoid seeing the following kind errors in your `release` builds:

```
PlatformException(startScan, Field androidScanMode_ for m0.e0 not found. Known fields are
 [private int m0.e0.q, private b3.b0$i m0.e0.r, private boolean m0.e0.s, private static final m0.e0 m0.e0.t,
 private static volatile b3.a1 m0.e0.u], java.lang.RuntimeException: Field androidScanMode_ for m0.e0 not found
```

closed 2023-08-04 by chipweinberger