chipweinberger/flutter_blue_plus #468: [Bug]: NPE in Google Play Store Pre-Launch report

MrCsabaToth, 2023-08-04:
### FlutterBluePlus Version

1.12.9

### Flutter Version

3.13.0-0.2.pre channel beta

### What OS?

Android

### OS Version

Android 9

### What happened?

I got a crash by the Pre-Launch report by the Google Play Store. It is a Pixel 2 device. I tried my release apk on an Android 13 device and it worked fine.

### Logs

```shell
Exception java.lang.NullPointerException: Attempt to invoke virtual method 'int android.bluetooth.BluetoothManager.getConnectionState(android.bluetooth.BluetoothDevice, int)' on a null object reference
  at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmScanResult (FlutterBluePlusPlugin.java:1707)
  at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$2.onScanResult (FlutterBluePlusPlugin.java:1235)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run (BluetoothLeScanner.java:531)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at androidx.test.espresso.base.Interrogator.loopAndInterrogate (Interrogator.java:10)
  at androidx.test.espresso.base.UiControllerImpl.loopUntil (UiControllerImpl.java:7)
  at androidx.test.espresso.base.UiControllerImpl.loopUntil (UiControllerImpl.java:1)
  at androidx.test.espresso.base.UiControllerImpl.loopMainThreadForAtLeast (UiControllerImpl.java:7)
  at androidx.test.espresso.action.Tap$1.sendTap (Tap.java:4)
  at androidx.test.espresso.action.GeneralClickAction.perform (GeneralClickAction.java:4)
  at androidx.test.espresso.ViewInteraction$SingleExecutionViewAction.perform (ViewInteraction.java:2)
  at androidx.test.espresso.ViewInteraction.doPerform (ViewInteraction.java:23)
  at androidx.test.espresso.ViewInteraction.-$$Nest$mdoPerform
  at androidx.test.espresso.ViewInteraction$1.call (ViewInteraction.java:6)
  at androidx.test.espresso.ViewInteraction$1.call (ViewInteraction.java:1)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:201)
  at android.os.Looper.loop (Looper.java:288)
  at android.app.ActivityThread.main (ActivityThread.java:7870)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:548)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1003)
```
```

MrCsabaToth, 2023-08-04:
I tried to repro with an Android 9 Moto G6 with the release APK and also with a debug version. When I do a scan the log is flooded with

    D/BluetoothManager(10375): getConnectionState()
    D/BluetoothManager(10375): getConnectedDevices

It scared me because in the past I removed the connection indicator from my app for the reason that it sometimes messed with the Bluetooth protocol stack in a long run when the widget was polling all the time. But this is unrelated.

MrCsabaToth, 2023-08-04:
the `com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$2.onScanResult (FlutterBluePlusPlugin.java:1235)` is

    // see BmScanResult
    HashMap<String, Object> rr = bmScanResult(device, result);

and then the `at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmScanResult (FlutterBluePlusPlugin.java:1707)` is

    // connection state
    int cs = mBluetoothManager.getConnectionState(device, BluetoothProfile.GATT);

If the mBluetoothManager is null then that's pretty catastrophic thing and this might be a quirk related to testing: those robots are dumb and maybe they click through themselves the Bluetooth permissions and Bluetooth might not be enabled. I'll test that case sometimes later.

chipweinberger, 2023-08-04:
`mBluetoothManager` is initialized here, if _any_ method is called.

```
    @Override
    public void onMethodCall(@NonNull MethodCall call,
                                 @NonNull Result result)
    {
        try {
            log(LogLevel.DEBUG, "[FBP-Android] onMethodCall: " + call.method);

            // initialize adapter
            if (mBluetoothAdapter == null) {
                log(LogLevel.DEBUG, "[FBP-Android] initializing BluetoothAdapter");
                mBluetoothManager = (BluetoothManager) this.context.getSystemService(Context.BLUETOOTH_SERVICE);
                mBluetoothAdapter = mBluetoothManager != null ? mBluetoothManager.getAdapter() : null;
            }
```

But then deinitialized here

```
    @Override
    public void onDetachedFromEngine(@NonNull FlutterPluginBinding binding)
    {
        Log.d(TAG, "onDetachedFromEngine");

        mBluetoothAdapter = null;
        mBluetoothManager = null;
```

chipweinberger, 2023-08-04:
@MrCsabaToth , is `onDetachedFromEngine` in the logs?

is `[FBP-Android] onMethodCall:` in the logs?

chipweinberger, 2023-08-04:
fixed in 1.12.12

closed 2023-08-04 by chipweinberger

MrCsabaToth, 2023-08-04:
I tried to repro this with FBP 1.12.9 and intentionally turning off the Bluetooth, but was not able to. When the app was started with BP off or when I turned off BT while the app was running.

Being BT off is not a crazy scenario BTW. I have it always on because of my WearOS watch, however many users may keep it off for battery saving or security reasons (less attack surface). On top of that I'm also distributing my app on Samsung Galaxy Store and [Huawei App Gallery](https://appgallery.huawei.com/) and the latter one requires the app to be able to start and gracefully behave when BT is off.