chipweinberger/flutter_blue_plus #469: [Bug]: iOS FlutterBluePlus.isAvailable throw error

lwd369, 2023-08-04:
### FlutterBluePlus Version

1.12.11

### Flutter Version

3.10.4

### What OS?

iOS

### OS Version

iOS 16.3.1

### What happened?

call `FlutterBluePlus.isAvailable` will throw `type 'int' is not a subtype of type 'FutureOr<bool>'`

### Logs

```shell
flutter: type 'int' is not a subtype of type 'FutureOr<bool>'
flutter: #0      FlutterBluePlus.isAvailable
flutter_blue_plus.dart:46
<asynchronous suspension>
#1      BleService._checkBluetoothStatus
ble_service.dart:344
<asynchronous suspension>
#2      BleService.scanDevices
ble_service.dart:95
```

chipweinberger, 2023-08-04:
regressed here https://github.com/boskokg/flutter_blue_plus/commit/a83ade69052b2d0794e1acc74fb85d12a0da07d3

chipweinberger, 2023-08-04:
fixed in 1.12.13

closed 2023-08-04 by chipweinberger