chipweinberger/flutter_blue_plus #470: [Bug]:

miaozhenkun, 2023-08-04:
### FlutterBluePlus Version

1.12.11

### Flutter Version

3.3.8

### What OS?

Android

### OS Version

13.0.1

### What happened?

android app crash

### Logs

```shell
D/FlutterGeolocator(18573): Unbinding from location service.
D/FlutterGeolocator(18573): Destroying location service.
D/FlutterGeolocator(18573): Stopping location service.
D/FlutterGeolocator(18573): Destroyed location service.
D/AndroidRuntime(18573): Shutting down VM
E/AndroidRuntime(18573): FATAL EXCEPTION: main
E/AndroidRuntime(18573): java.lang.NullPointerException: Attempt to invoke virtual method 'int android.bluetooth.BluetoothManager.getConnectionState(android.bluetooth.BluetoothDevice, int)' on a null object reference
E/AndroidRuntime(18573): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmScanResult(FlutterBluePlusPlugin.java:1717)
E/AndroidRuntime(18573): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$2.onScanResult(FlutterBluePlusPlugin.java:1245)
E/AndroidRuntime(18573): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:570)
E/AndroidRuntime(18573): 	at android.os.Handler.handleCallback(Handler.java:942)
E/AndroidRuntime(18573): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime(18573): 	at android.os.Looper.loopOnce(Looper.java:223)
E/AndroidRuntime(18573): 	at android.os.Looper.loop(Looper.java:324)
E/AndroidRuntime(18573): 	at android.app.ActivityThread.main(ActivityThread.java:8486)
E/AndroidRuntime(18573): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(18573): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:582)
E/AndroidRuntime(18573): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:996)
W/libc    (18573): Access denied finding property "vendor.cellular.ril.imsi"
```

chipweinberger, 2023-08-04:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/468

chipweinberger, 2023-08-04:
fixed in 1.12.12

closed 2023-08-04 by chipweinberger