chipweinberger/flutter_blue_plus #472: [Question]:

NewAge23, 2023-08-04:
### Ask your question

I/flutter (17980): Error writing data: FlutterBluePlusException: name:writeCharacteristic errorCode:13, errorString:GATT_INVALID_ATTRIBUTE_LENGTH if i am sendning one value its okay its send it, but if i send [0x00,0xff,0xff,0xff]this value is provide error please provide solution

chipweinberger, 2023-08-04:
[image]

chipweinberger, 2023-08-04:
your peripheral device does not support 4 bytes. It's too long.

Your peripheral device is probably improperly configured.

closed 2023-08-04 by chipweinberger

NewAge23, 2023-08-04:
Android: writeCharacteristic error GATT_INVALID_ATTRIBUTE_LENGTH but my peripheral device support 4 bytes. Also when the bluetooth device wa send the output in hterm also but now in updated plugin it does not work

chipweinberger, 2023-08-04:
This error comes from Android. 

https://developer.android.com/reference/android/bluetooth/BluetoothGatt#GATT_INVALID_ATTRIBUTE_LENGTH

It is unlikely to be a FlutterBluePlus problem. It is android or your device problem.

NewAge23, 2023-08-04:
Future<void> _writeData() async {
    List<int> dataToWrite = [0x51,0x32];
    final int maxChunkSize =
        20; // Replace with your characteristic's maximum allowed length

    try {
      for (int i = 0; i < dataToWrite.length; i += maxChunkSize) {
        final chunk = dataToWrite.sublist(
            i,
            i + maxChunkSize > dataToWrite.length
                ? dataToWrite.length
                : i + maxChunkSize);

        await widget.characteristic.write(chunk,
            withoutResponse:
                widget.characteristic.properties.writeWithoutResponse);

        // Add a delay if needed between chunks to avoid overloading the BLE stack.
        // await Future.delayed(Duration(milliseconds: 100));
      }

      setState(() {
        // Update the UI or state if needed after the write operation
      });

      print("Data written successfully: $dataToWrite");
    } catch (e) {
      print("Error writing data: $e");
      // Handle the error, show a snackbar, or display an error message to the user.
    }
  }this is my code can you please correct my code for 4 bytes

chipweinberger, 2023-08-04:
```
 Future<void> _writeData() async {
    List<int> dataToWrite = [0x51,0x32];
    final int maxChunkSize =
        20; // Replace with your characteristic's maximum allowed length

    try {
      for (int i = 0; i < dataToWrite.length; i += maxChunkSize) {
        final chunk = dataToWrite.sublist(
            i,
            i + maxChunkSize > dataToWrite.length
                ? dataToWrite.length
                : i + maxChunkSize);
            
        try {
            await widget.characteristic.write(chunk,
                withoutResponse: widget.characteristic.properties.writeWithoutResponse);
        } catch(FlutterBluePlusException e) {
            if(e.errorString == "GATT_INVALID_ATTRIBUTE_LENGTH") {
                 //ignore
            } else {
                rethrow;
            }
        };

        // Add a delay if needed between chunks to avoid overloading the BLE stack.
        // await Future.delayed(Duration(milliseconds: 100));
      }

      setState(() {
        // Update the UI or state if needed after the write operation
      });

      print("Data written successfully: $dataToWrite");
    } catch (e) {
      print("Error writing data: $e");
      // Handle the error, show a snackbar, or display an error message to the user.
    }
  }
```

NewAge23, 2023-08-04:
Future<void> _writeData() async {
    List<int> dataToWrite = [0x00, 0xff, 0xff, 0xff];
    final int maxChunkSize =
        20; // Replace with your characteristic's maximum allowed length

    try {
      for (int i = 0; i < dataToWrite.length; i += maxChunkSize) {
        final chunk = dataToWrite.sublist(
            i,
            i + maxChunkSize > dataToWrite.length
                ? dataToWrite.length
                : i + maxChunkSize);

        await widget.characteristic.write(chunk,
            withoutResponse:
                widget.characteristic.properties.writeWithoutResponse);

        try {
          await widget.characteristic.write(chunk,
              withoutResponse:
                  widget.characteristic.properties.writeWithoutResponse);
        } on FlutterBluePlusException catch (e) {
          if (e.errorString == "GATT_INVALID_ATTRIBUTE_LENGTH") {
            //ignore
          } else {
            rethrow;
          }
        }
        ;

        // Add a delay if needed between chunks to avoid overloading the BLE stack.
        // await Future.delayed(Duration(milliseconds: 100));
      }

      setState(() {
        // Update the UI or state if needed after the write operation
      });

      print("Data written successfully: $dataToWrite");
    } catch (e) {
      print("Error writing data: $e");
      // Handle the error, show a snackbar, or display an error message to the user.
    }
  }
it seems provde this error
I/flutter (24537): Error writing data: FlutterBluePlusException: name:writeCharacteristic errorCode:13, errorString:GATT_INVALID_ATTRIBUTE_LENGTH

chipweinberger, 2023-08-04:
You have mistake. You have this code 2x. Delete first one.

```
    await widget.characteristic.write(chunk,
        withoutResponse:
            widget.characteristic.properties.writeWithoutResponse);
```

NewAge23, 2023-08-04:
D/[FBP-Android](24537): [FBP-Android] onCharacteristicWrite: uuid: 0000fff1-0000-1000-8000-00805f........ status: 13
I/flutter (24537): Data written successfully: [0, 255, 255, 255] 

it provide status 13 it means it does not to the bluetooth device that message. please provide correct solution

chipweinberger, 2023-08-04:
ignore the error.

NewAge23, 2023-08-04:
is that means my data is succesfully sent to my bluetooth device?

MrCsabaToth, 2023-08-04:
> is that means my data is succesfully sent to my bluetooth device?

What does the device do? Does it act like the setting was successful?

NewAge23, 2023-08-05:
when i was used that  plugin older version . it communicate to the hterm software but  now in upgraded version of plugin it does not communicate to the hterm what i can do