chipweinberger/flutter_blue_plus #473: [Bug]: locateGatt failed problem

umuttsaltan, 2023-08-04:
### FlutterBluePlus Version

1.12.13

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 11

### What happened?

BLE Device connection step by step :

Scan devices
Find the specific device from remoteId
Connect and then discoverServices
I can see from BLE device, connection is successful but discoverServices not working.

I can share my code from discoverServices part if needed.

### Logs

```shell
e_id: 66:C8:62:F1:25:73, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -68, advertisement_data: {connectable: 1, manufacturer_data: {76: 10050598c009e0}, tx_power_level: 24, service_data: {}}, device: {remote_id: 62:F7:2E:2F:8B:FF, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -57, advertisement_data: {connectable: 0, manufacturer_data: {6: 01092022b88a2fdbf033a0a37f8021b2533d55deee3cdce98a3077}, service_data: {}}, device: {remote_id: 0E:CC:0A:98:20:63, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -57, advertisement_data: {connectable: 0, manufacturer_data: {6: 0109200246ae3091f49622fc504059e66e879ec317ffe8992ebfa6}, service_data: {}}, device: {remote_id: 46:32:6A:D3:31:4B, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -64, advertisement_data: {connectable: 1, manufacturer_data: {76: 1002350c0f058035ade4d2}, tx_power_level: 8, service_data: {}}, device: {remote_id: 61:53:5C:07:FB:3F, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -60, advertisement_data: {connectable: 1, manufacturer_data: {76: 0c0e00e3737cc47c0de05b1abe9f46861005111c925c5b}, service_data: {}}, device: {remote_id: 4C:CE:D2:D6:8A:81, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -54, advertisement_data: {connectable: 0, manufacturer_data: {6: 010920220e32dfe6db332fa844e855d258a8ccc78c096fb430cd32}, service_data: {}}, device: {remote_id: 15:E1:66:2C:8A:8D, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -64, advertisement_data: {connectable: 0, manufacturer_data: {6: 010f2002d9bf09866cfe855b02100f38bc83aee390a2928701b3e3}, service_data: {}}, device: {remote_id: 40:1D:E5:B8:11:C6, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -59, advertisement_data: {connectable: 0, manufacturer_data: {76: 090603380a00003c}, service_data: {}}, device: {remote_id: 12:DA:A4:31:3E:25, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -63, advertisement_data: {connectable: 0, manufacturer_data: {6: 01092002a667eb24ae20415c4d387cc0d753aa5f39d06caf1e6c80}, service_data: {}}, device: {remote_id: 05:A8:4B:57:46:86, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -63, advertisement_data: {connectable: 0, manufacturer_data: {6: 0109210a3a637aadde384d4b532d415247452d3135}, service_data: {}}, device: {remote_id: 3C:84:8A:8A:67:FC, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -56, advertisement_data: {connectable: 0, manufacturer_data: {6: 01092022b88a2fdbf033a0a37f8021b2533d55deee3cdce98a3077}, service_data: {}}, device: {remote_id: 0E:CC:0A:98:20:63, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -51, advertisement_data: {connectable: 0, manufacturer_data: {76: 090603960a00003f}, service_data: {}}, device: {remote_id: 2B:83:71:0E:15:17, type: 0}}}
I/flutter (15357): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] <connect> result: null
I/flutter (15357): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -42, advertisement_data: {connectable: 1, manufacturer_data: {76: 10063a1e83590522}, tx_power_level: 8, service_data: {}}, device: {remote_id: 5E:B1:83:70:D8:3B, type: 0}}}
[... 146 lines cut ...]
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <getConnectedSystemDevices> result: {devices: [{remote_id: 60:55:F9:85:59:2A, type: 2, local_name: PLUG_CHARGER}]}
I/flutter (15467): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <connect> result: null
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <connect> result: null
I/flutter (15467): [FBP] <connect> result: null
I/flutter (15467): [FBP] <getConnectedSystemDevices> result: {devices: [{remote_id: 60:55:F9:85:59:2A, type: 2, local_name: PLUG_CHARGER}]}
I/flutter (15467): [FBP] <connect> args: {remote_id: 60:55:F9:85:59:2A, auto_connect: 0}
I/flutter (15467): [FBP] <connect> result: null
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (15467): [FBP] <connect> result: null
I/flutter (15467): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
```

chipweinberger, 2023-08-04:
1.  why do you call `discoverServices` so many times?
2. does your `PLUG_CHARGER` actually have services?

umuttsaltan, 2023-08-04:
1. I don't, just one time when i connect the device but in log output it looks like trying many times.
2. Yeah it has, i can connect to same services with IOS device.
3. After log output exceptions comes and discoverServices doesnt timeout. 

There are some logs with exceptions 

```dart
I/flutter (18337): Bağlantı hatası: PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1038)
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:477)
I/flutter (18337):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (18337):      at android.os.Handler.handleCallback(Handler.java:938)
I/flutter (18337):      at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (18337):      at android.os.Looper.loop(Looper.java:223)
I/flutter (18337):      at android.app.ActivityThread.
I/flutter (18337): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (18337): Bağlantı hatası: PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1038)
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:477)
I/flutter (18337):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (18337):      at android.os.Handler.handleCallback(Handler.java:938)
I/flutter (18337):      at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (18337):      at android.os.Looper.loop(Looper.java:223)
I/flutter (18337):      at android.app.ActivityThread.
I/flutter (18337): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (18337): Bağlantı hatası: PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1038)
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:477)
I/flutter (18337):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (18337):      at android.os.Handler.handleCallback(Handler.java:938)
I/flutter (18337):      at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (18337):      at android.os.Looper.loop(Looper.java:223)
I/flutter (18337):      at android.app.ActivityThread.
I/flutter (18337): Bağlantı hatası: PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1038)
I/flutter (18337):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:477)
I/flutter (18337):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (18337):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (18337):      at android.os.Handler.handleCallback(Handler.java:938)
I/flutter (18337):      at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (18337):      at android.os.Looper.loop(Looper.java:223)
[... cut ...]
I/flutter (18337): [FBP] <connect> result: null
I/flutter (18337): [FBP] <stopScan> result: null
I/flutter (18337): [FBP] <stopScan> result: null
I/flutter (18337): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
I/flutter (18337): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -59, advertisement_data: {connectable: 0, manufacturer_data: {6: 01092002dbcb58272a41695d531e94f2c2f7281221e9b048aad64c}, service_data: {}}, device: {remote_id: 26:B5:4A:BC:06:70, type: 0}}}
I/flutter (18337): [FBP] <discoverServices> args: 60:55:F9:85:59:2A
```

chipweinberger, 2023-08-04:
you are still missing important logs.

1. does onDiscoverServices ever get called in the logs?
2. does the device lose connection before you call discoverServices?

umuttsaltan, 2023-08-07:
1. It doesnt call 'onDiscoverServices', i checked it many times. 
2. No, still there is a connection between device and app.

Sorry, but i dont know how to get more logs. Can u describe me?

chipweinberger, 2023-08-07:
This may be due to FlutterBluePlus not considering bond state after connection.

https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07

[image]

chipweinberger, 2023-08-08:
I've release 1.13.0, which handles bonded devices.

Reopen a new issue if you still have problems.

closed 2023-08-08 by chipweinberger