chipweinberger/flutter_blue_plus #474: [Feature]: Is there a way to get the pairing request result? i.e. pairing code was wrong

biklas7, 2023-08-04:
### Feature Request

First, I just want to thank the maintainers of this package for all the amazing work and for being so active 🙏 it's really good to see a proper Bluetooth package for Flutter 👍 

I'm connecting to a device that requires the user to enter a pairing code. Is there a way to know if the pairing request was successful? I mean if the user clicked on Pair or Cancel or entered the wrong code?

What happens is that when calling `device.connect( ... )` the user sees the pairing pop up to enter the code, but before the pairing is finished I've noticed that `device.connectionState` stream emits `connected`.

So there's no real way of knowing if the pairing request was successful.

What I've noticed is that if the user enters the wrong pairing code for example, the `device.connectionState` stream emits `connected` but then emits `disconnected`.

I know that on Android we have [ACTION_BOND_STATE_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#ACTION_BOND_STATE_CHANGED), not sure if there's something similar on iOS.

Also noticed, and correct me if I'm wrong, that calling `device.pair()` doesn't really wait for the actual bonding state result, because I didn't see any implementation that registers for [ACTION_BOND_STATE_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#ACTION_BOND_STATE_CHANGED) intents to be notified when the bonding process completes. It just uses the returned bool of calling `createBond()` which according to the docs returns `false on immediate error, true if bonding will begin` so it doesn't really mean that the device is BONDED.

So the bottom line is, would it be possible to expose a stream called `bondingState` (or something else) similar to the `connectionState` stream, but instead it would emit the actual device bonding state using [ACTION_BOND_STATE_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#ACTION_BOND_STATE_CHANGED) on Android and the equivalent, if available, on iOS?

chipweinberger, 2023-08-04:
This is not work I plan to do, but feel free to open a PR.

Make sure you follow the existing code conventions & handle all errors. The code must be clearly right.

AymenXD, 2023-08-04:
Thank you for raising this issue. Please keep us updated if there are any news or workarounds concerning this matter.

Keep up the good work. Best BLE package so far!!!

chipweinberger, 2023-08-08:
I've started work supporting it, on this branch: https://github.com/boskokg/flutter_blue_plus/tree/improve-android-bond-support

This blog post is extremely useful:
https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442

Bonding has a few annoyances:
- you must wait until bonded before you call discoverServices(), read(), write(), etc
- you can "lose" a bond, where you must disconnect & reconnect
- some devices need extra delays (again, check blog post)

On my branch, there are still a lot of decisions left:
- what should the public API be? 
- how can we make the user wait until bonded before calling  discoverServices(), read(), write(), etc?
- how should we handle losing a bond?

chipweinberger, 2023-08-08:
> Is there a way to get the pairing result? i.e. pairing code was wrong

You can try to read an encrypted characteristic. If you get an INSUFFICIENT_AUTHENTICATION error than you know pairing failed.

chipweinberger, 2023-08-08:
Okay, I've released `1.13.0`, which returns the pairing result.

Reopen a new issue if you have problems.

closed 2023-08-08 by chipweinberger

biklas7, 2023-08-08:
@chipweinberger Amazing work! Thanks again 👍