chipweinberger/flutter_blue_plus #475: [Question]: it did not read the data to the hterm  software

NewAge23, 2023-08-04:
### Ask your question

: it did not read the data to the hterm  software

MrCsabaToth, 2023-08-04:
We won't be able to help unless you tell more details. What plugin version, what platform, what device, what logs do you have, was it an exception or something else, what is your source code - to begin with.

MrCsabaToth, 2023-08-04:
Is this #472 related?

NewAge23, 2023-08-05:
Future<void> _sendMessage() async {
    List<int> messageBytes = [0x00, 0xff, 0xff, 0xff, 0x57];

    try {
      List<BluetoothService> services = await widget.device.discoverServices();
      for (BluetoothService service in services) {
        if (service.uuid == Guid('0000fff0-0000-1000-8000-00805f9b34fb')) {
          List<BluetoothCharacteristic> characteristics =
              service.characteristics;
          for (BluetoothCharacteristic characteristic in characteristics) {
            // if (characteristic.uuid ==
            //     Guid('0000fff1-0000-1000-8000-00805f9b34fb')) {
            if (characteristic.properties.write) {
              // Use a fixed chunk size (adjust as needed)
              const chunkSize = 1;
              for (int i = 0; i < messageBytes.length; i += chunkSize) {
                int end = i + chunkSize;
                if (end > messageBytes.length) {
                  end = messageBytes.length;
                }
                final chunk = messageBytes.sublist(i, end);
                await characteristic.write(chunk);
                print('Chunk sent: ${chunk}..............write data...');
              }
              print('Message sent: ${messageBytes}............input data....');
              return;
            }
            // }
          }
        }
      }

      print('Desired characteristic not found.');
    } catch (e) {
      print('Error sending message: $e');
    }
  } i tried so much but i want to send that message in one time but if i am sending that data it goes throw with ony by one

chipweinberger, 2023-08-08:
Sorry, it will be too difficult to help you further with this. It seems like a device problem. You will need to debug it further yourself.

closed 2023-08-08 by chipweinberger