chipweinberger/flutter_blue_plus #477: [Bug]: NoSuchMethodError on Android 6.0 - 4.4 (API level 23-19)

MrCsabaToth, 2023-08-06:
### FlutterBluePlus Version

1.12.13

### Flutter Version

3.13.0-0.2.pre

### What OS?

Android

### OS Version

6.0.1

### What happened?

I tested the plugin on an Android 6.0.1 device and faced an exception. The exception lead to this issue: https://stackoverflow.com/questions/58408832/nosuchmethoderror-no-interface-method-getordefaultljava-lang-objectljava-lang

I have a user base with Android 6 and older (Android 5, 4.4) devices. I'll craft a PR.
Ref https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/451

### Logs

```shell
E/AndroidRuntime(16712): FATAL EXCEPTION: main
E/AndroidRuntime(16712): Process: dev.csaba.track_my_indoor_exercise, PID: 16712
E/AndroidRuntime(16712): java.lang.NoSuchMethodError: No virtual method getOrDefault(Ljava/lang/Object;Ljava/lang/Object;)Ljava/lang/Object; in class Ljava/util/HashMap; or its super classes (declaration of 'java.util.HashMap' appears in /system/framework/core-libart.jar)
E/AndroidRuntime(16712): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.fetchFilters(FlutterBluePlusPlugin.java:1189)
E/AndroidRuntime(16712): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:298)
E/AndroidRuntime(16712): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/AndroidRuntime(16712): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/AndroidRuntime(16712): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/AndroidRuntime(16712): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(D8$$SyntheticClass)
E/AndroidRuntime(16712): 	at android.os.Handler.handleCallback(Handler.java:739)
E/AndroidRuntime(16712): 	at android.os.Handler.dispatchMessage(Handler.java:95)
E/AndroidRuntime(16712): 	at android.os.Looper.loop(Looper.java:148)
E/AndroidRuntime(16712): 	at android.app.ActivityThread.main(ActivityThread.java:5480)
E/AndroidRuntime(16712): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(16712): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
E/AndroidRuntime(16712): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
I/Process (16712): Sending signal. PID: 16712 SIG: 9
```

chipweinberger, 2023-08-07:
lets just not use getOrDefault.

`        int macCount = scanSettings.get("mac_count") != null ? (int) scanSettings.get("mac_count") : 0;`

chipweinberger, 2023-08-07:
ill push

closed 2023-08-07 by chipweinberger

chipweinberger, 2023-08-07:
fixed 53452ac1f135af18854f1cb06cd52fee8f864f5e