chipweinberger/flutter_blue_plus #479: [Bug]: Java code has some calls which are not supported below API level 21

MrCsabaToth, 2023-08-06:
### FlutterBluePlus Version

1.12.13

### Flutter Version

3.13.0-0.2.pre

### What OS?

Android

### OS Version

6.0.1

### What happened?

While investigating #477 I saw that there are some instructions which are supported at API level 21 (Android 5), but not at API level 19 or 20 (Android 4.4). I have users using such devices, so it'd be good to see what the original Flutter Blue plugin does, I most definitely not want to raise the min SDK level.

### Logs

```shell
Here are the calls:
* Call requires API level 21 (current min is 19): `android.bluetooth.BluetoothGatt#requestMtu` - I think we can live without modifying this, MTU size request just won't be supported on API level 20 or 19, not a big whop
* Call requires API level 21 (current min is 19): `android.bluetooth.BluetoothGatt#requestConnectionPriority` - same as with MTU size request, not a big whop, we can just document it
* Call requires API level 26 (current min is 19): `android.bluetooth.BluetoothGatt#setPreferredPhy` - same as previous two: this is also a separate embedding method call, can be simply documented that it won't work on API level below 26, not a big whop
* Call requires API level 21 (current min is 19): `android.bluetooth.BluetoothAdapter#getBluetoothLeScanner`
* Call requires API level 21 (current min is 19): `new android.bluetooth.le.ScanSettings.Builder`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.ScanSettings.Builder#setScanMode`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.ScanSettings.Builder#build`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.BluetoothLeScanner#startScan`
* Call requires API level 21 (current min is 19): `android.bluetooth.BluetoothAdapter#getBluetoothLeScanner`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.BluetoothLeScanner#stopScan`
* Call requires API level 21 (current min is 19): `new android.bluetooth.le.ScanFilter.Builder`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.ScanFilter.Builder#setDeviceAddress`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.ScanFilter.Builder#build`
* Call requires API level 21 (current min is 19): `android.bluetooth.le.ScanFilter.Builder#setServiceUuid`

I think the last 4 issues: I'd be OK to simply shortcut and not support filtering on API level 19 and 20. For the `ScanSettings` and rest we can also just skip those if possible?
```

chipweinberger, 2023-08-07:
> The Android KitKat is the 4.4 release version. Was publicly released on September 3, 2013.

Wow so old. 0.5% of users are on that version. We should just bump the minSdk.

[image]

MrCsabaToth, 2023-08-07:
I'm willing to put some work in order to not lose this, can you point me to some commits when we lost the support?

chipweinberger, 2023-08-07:
afaict we never supported 19

`BluetoothLeScanner` requires 21.

chipweinberger, 2023-08-07:
Nevermind, it used to work like this

https://github.com/pauldemarco/flutter_blue/blob/master/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java

```
    private void startScan18(Protos.ScanSettings proto) throws IllegalStateException {
        List<String> serviceUuids = proto.getServiceUuidsList();
        UUID[] uuids = new UUID[serviceUuids.size()];
        for(int i = 0; i < serviceUuids.size(); i++) {
            uuids[i] = UUID.fromString(serviceUuids.get(i));
        }
        boolean success = mBluetoothAdapter.startLeScan(uuids, getScanCallback18());
        if(!success) throw new IllegalStateException("getBluetoothLeScanner() is null. Is the Adapter on?");
    }
```

chipweinberger, 2023-08-08:
Closing as not planned.

A lot of feature work is required for this: https://github.com/boskokg/flutter_blue_plus/issues/474

closed 2023-08-08 by chipweinberger

MrCsabaToth, 2023-08-08:
It also surprises me, but by Google Play Store report I have about 1-1.3% user base who use such ancient Android.

chipweinberger, 2023-11-09:
btw, Flutter itself only has "best effort" support for sdk 19 - 20

https://docs.flutter.dev/reference/supported-platforms

so it seems these sdks are gonna be pretty bad experience regardless.

MrCsabaToth, 2023-11-10:
I got over this a while ago. At some point I wanted to work on it, but 1.) I had to admit I don't have time 2.) I don't think it'd worth the effort. The newer Bluetooth API means newer BT tech stack at the Android OS level as well, and that's beneficial for everyone. Connections can be more stable and a ton of improvements (of the BT protocol stack), too long to list.