chipweinberger/flutter_blue_plus #482: [Bug]:  App crash

miaozhenkun, 2023-08-07:
### FlutterBluePlus Version

1.12.13

### Flutter Version

3.3.8

### What OS?

Android

### OS Version

13.0.1

### What happened?

Thank you for your continuous contribution. I encountered a bug where I proactively connected a Bluetooth accessory device in the Android system and then returned to the app. Use Scan Bluetooth and connect Bluetooth. The app may crash and restart. Unfortunately, I couldn't find the reason.

### Logs

```shell
I couldn't find the reason
```

chipweinberger, 2023-08-08:
Thanks for the report. If you can find the logs, please reopen.

Until then, it is impossible to know if this is a FBP issue.

closed 2023-08-08 by chipweinberger

miaozhenkun, 2023-08-08:
Sorry, it's not a plugin bug, it should be an iQOO phone bug in the Android 13 system.