chipweinberger/flutter_blue_plus #484: [Question]: I am receiving a Timeout Exception whenever i want to send commands to my ble

DanielMbalyo, 2023-08-07:
### Ask your question

The functionality is as follows: 

Future<String> _sendCommand(String command) async {
    // List<int> commandData = utf8.encode(command);
    String response = '';
    final math = Random();
    List<int> commandData = [math.nextInt(255), math.nextInt(255), math.nextInt(255), math.nextInt(255)];
    const int maxChunkSize = 20; 
    Guid serviceUuid = Guid('00001800-0000-1000-8000-00805f9b34fb');
    Guid characteristicUuid = Guid('00002a00-0000-1000-8000-00805f9b34fb');
    List<BluetoothService> services = await _connDevice!.discoverServices();
    BluetoothService? service = services.firstWhereOrNull((service) => service.uuid == serviceUuid);
    if (service != null) {
      if (service.uuid == serviceUuid) {
        List<BluetoothCharacteristic> characteristics = service.characteristics;
        BluetoothCharacteristic? characteristic = characteristics.firstWhereOrNull(
          (characteristic) => characteristic.uuid == characteristicUuid);
        if (characteristic != null) {
          if (characteristic.uuid == characteristicUuid) {
            for (int i = 0; i < commandData.length; i += maxChunkSize) {
              final chunk = commandData.sublist(i, i + maxChunkSize > commandData.length ? commandData.length : i + maxChunkSize);
              try {
                await characteristic.write(chunk, withoutResponse: characteristic.properties.writeWithoutResponse);
              } on FlutterBluePlusException catch (e) {
                if (e.errorString == "GATT_INVALID_ATTRIBUTE_LENGTH") {
                  //ignore
                } else {
                  rethrow;
                }
              }
              break;
            }
            await characteristic.setNotifyValue(true);
            await characteristic.lastValueStream.listen((data) {
              response += utf8.decode(data);
            }).asFuture();
          }
        }
      }
    }
    await Future.delayed(const Duration(milliseconds: 100), () => print('Received response: $response'));
    return response;
  }

I am using flutter_blue_plus: 1.12.13, thank you in advance

chipweinberger, 2023-08-07:
what does your **connect** code look like?

DanielMbalyo, 2023-08-07:
`Future _connectDevice(List<String> commands) async {
    if (_connDevice == null) return;
    print(commands);
    try {
      List<String> responses = [];
      await _connDevice!.connect();
      await _connDevice!.requestMtu(64);
      for (String command in commands) {
        String res = await _sendCommand(command);
        if (res.isNotEmpty) responses.add(res);
      }
      print(responses);
      await _sendAcknowledgement(responses);
    } catch (exception) {
      throw Exception('Cannot connect, exception occurred: $exception');
    } finally {
      await _connDevice!.disconnect();
    }
  }`

DanielMbalyo, 2023-08-07:
```
Future _connectDevice(List<String> commands) async {
    if (_connDevice == null) return;
    print(commands);
    try {
      List<String> responses = [];
      await _connDevice!.connect();
      await _connDevice!.requestMtu(64);
      for (String command in commands) {
        String res = await _sendCommand(command);
        if (res.isNotEmpty) responses.add(res);
      }
      print(responses);
      await _sendAcknowledgement(responses);
    } catch (exception) {
      throw Exception('Cannot connect, exception occurred: $exception');
    } finally {
      await _connDevice!.disconnect();
    }
  }
```

chipweinberger, 2023-08-07:
[image]

For some reason your device disconnects and has trouble reconnecting. Your device could have crashed.

I don't see any obvious bugs in FlutterBluePlus.

chipweinberger, 2023-08-07:
`unregisterApp` is the problem. I don't know what that is, but googling for it returns similar errors.

[image]

chipweinberger, 2023-08-09:
There have been improvements made in 1.14.0.

Please open a new issue with new logs if you hit this issue again.

closed 2023-08-09 by chipweinberger

DanielMbalyo, 2023-08-11:
Thanks, the problem was happening because I was writting to a wrong characteristics.

chipweinberger, 2023-08-11:
```
      for (String command in commands) {
        String res = await _sendCommand(command);
        if (res.isNotEmpty) responses.add(res);
      }
    } catch (exception) {
      throw Exception('Cannot connect, exception occurred: $exception');
```

Ah I see the issue now, You are logging "Cannot connect" but the issue was actually in _sendCommand. Yes, that is confusing.