chipweinberger/flutter_blue_plus #485: Services is not working properly[Question]:

NewAge23, 2023-08-08:
### Ask your question

Discover Service not working properly

Xu-158, 2023-08-08:
I encountered the same problem on my Android phone，version 1.13.3

shawsa-artc, 2023-08-08:
same here, this author breaks the code in every version update

jagiro, 2023-08-08:
I tested a little bit and the version 1.12.14 was the last one that works fine on IOS and Android.

On 1.30.x on Android fails on `discoverServices()`. Don't show logs or errors to attach

chipweinberger, 2023-08-08:
logs are needed. I assume android?

chipweinberger, 2023-08-08:
discoverServices never returns

```
I/ue_plus_example( 8379): Compiler allocated 4413KB to compile void android.view.ViewRootImpl.performTraversals()
D/permissions_handler( 8379): No permissions found in manifest for: []3
D/permissions_handler( 8379): No permissions found in manifest for: []15
D/permissions_handler( 8379): Bluetooth permission missing in manifest
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android]( 8379): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 8379): isLeEnabled(): ON
D/BluetoothLeScanner( 8379): could not find callback wrapper
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter( 8379): isLeEnabled(): ON
D/BluetoothLeScanner( 8379): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 8379): isLeEnabled(): ON
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getAdapterState
4
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: connect
D/BluetoothGatt( 8379): connect() - device: 08:66:98:D6:8F:6C, auto: false
D/BluetoothGatt( 8379): registerApp()
D/BluetoothGatt( 8379): registerApp() - UUID=b2e5d3c9-b478-4894-a7a9-36b406ec0847
D/BluetoothGatt( 8379): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt( 8379): onClientConnectionState() - status=0 clientIf=6 device=08:66:98:D6:8F:6C
D/[FBP-Android]( 8379): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 8379): onConnectionUpdated() - Device=08:66:98:D6:8F:6C interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 8379): onConnectionUpdated() - Device=08:66:98:D6:8F:6C interval=36 latency=0 timeout=500 status=0
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: discoverServices
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 8379): [FBP-Android] onMethodCall: getConnectionState
```

chipweinberger, 2023-08-08:
fixed in 1.13.4

closed 2023-08-08 by chipweinberger