chipweinberger/flutter_blue_plus #486: [Bug]:

baconcheese113, 2023-08-09:
### FlutterBluePlus Version

1.12.13

### Flutter Version

3.7.12

### What OS?

Android

### OS Version

Android 9

### What happened?

It seems like if the services on a previously connected device change (as they might during an OTA firmware update) without closing the app, the device will throw an error trying to read from one of it's services on reconnect.

```dart
    print("in getChar()");
    final existingServices = d.servicesList ?? [];
    print("servicesList ${d.servicesList} and servicesStream.first ${d.servicesStream.first}");
    final services = existingServices.isNotEmpty ? existingServices : await d.discoverServices();
    print("services $services");
    final service = services.firstWhereOrNull((s) => s.uuid == Guid(serviceUuid));
    print("service $service");
    final char = service?.characteristics.firstWhereOrNull((c) => c.uuid == Guid(charUuid));
    print("char ${char?.lastValue}");
    print(">>> Got char");
    List<int> sensorVerBytes = await char!.read();
```

### Logs

```shell
I/flutter (15777): in getChar()
I/flutter (15777): servicesList [BluetoothService{remoteId: 80:EA:CA:70:00:01, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: 80:EA:CA:70:00:01, serviceUui
d: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a00-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutRespo
nse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remo
teId: 80:EA:CA:70:00:01, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a01-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, ext
I/flutter (15777): services [BluetoothService{remoteId: 80:EA:CA:70:00:01, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: 80:EA:CA:70:00:01, serviceUuid: 0
0001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a00-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse:
 false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId
: 80:EA:CA:70:00:01, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a01-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extende
I/flutter (15777): service BluetoothService{remoteId: 80:EA:CA:70:00:01, serviceUuid: 0000180a-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: 80:EA:CA:70:00:01, serviceUuid: 000
0180a-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a29-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: f
alse, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: 
80:EA:CA:70:00:01, serviceUuid: 0000180a-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a24-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedP
I/flutter (15777): char [48, 46, 49, 46, 51, 0]
I/flutter (15777): >>> Got char
D/[FBP-Android](15777): [FBP-Android] onMethodCall: readCharacteristic
E/flutter (15777): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.Exception: service not found on this device 
E/flutter (15777): service: 0000180a-0000-1000-8000-00805f9b34fb, java.lang.Exception: service not found on this device
E/flutter (15777): service: 0000180a-0000-1000-8000-00805f9b34fb
E/flutter (15777):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateCharacteristic(FlutterBluePlusPlugin.java:1052)
E/flutter (15777):      at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:499)
E/flutter (15777):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/flutter (15777):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter (15777):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter (15777):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter (15777):      at android.os.Handler.handleCallback(Handler.java:873)
E/flutter (15777):      at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (15777):      at android.os.Looper.loop(Looper.java:214)
E/flutter (15777):      at android.app.ActivityThread.main(ActivityThread.java:7073)
E/flutter (15777):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (15777):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
E/flutter (15777):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
E/flutter (15777): , null)
E/flutter (15777): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter (15777): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter (15777): <asynchronous suspension>
E/flutter (15777): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:267:19)
E/flutter (15777): <asynchronous suspension>
E/flutter (15777): #3      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:92:7)
E/flutter (15777): <asynchronous suspension>
E/flutter (15777): #4      _SensorUpdaterState.autoConnect (package:handle_it/feed/updaters/sensor_updater.dart:87:32)
E/flutter (15777): <asynchronous suspension>
E/flutter (15777):
D/BluetoothGatt(15777): onConnectionUpdated() - Device=80:EA:CA:70:00:01 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(15777): onConnectionUpdated() - Device=80:EA:CA:70:00:01 interval=39 latency=0 timeout=500 status=0
```

chipweinberger, 2023-08-09:
This seems expected, no?

You are getting an error thrown because you are trying to read a characteristic that does not exist anymore.

baconcheese113, 2023-08-09:
I guess I would expect that any services I find on the `services` array would be valid services. The service that it throws an error about is available both before and after the update, it's an [Assigned Number](https://btprodspecificationrefs.blob.core.windows.net/assigned-numbers/Assigned%20Number%20Types/Assigned_Numbers.pdf) which you can see on page 68 as 0x180a. This works *mostly* as-is with 1.4.0, so maybe I missed a change to the logic here?

chipweinberger, 2023-08-09:
You mean the `servicesList`?  `servicesList` did not exist in 1.4.0

> The service that it throws an error about **is available both before and after the update**

Oh that is an important detail!  I thought the service was gone.

Services and characteristics are just UUIDs, and if they are valid they should "just work", just like before.  

Perhaps you can investigate further. There are no changes that I know of. We get the service like this, just like in 1.4.0 i beleive.

```
        BluetoothGattService primaryService = gatt.getService(UUID.fromString(serviceId));

        if(primaryService == null) {
            throw new Exception("service not found on this device \n" +
                "service: "+ serviceId);
        }
```

baconcheese113, 2023-08-09:
[Here's](https://github.com/baconcheese113/handle-it-app/blob/b8211f46bf8282ec660c8357d7897cf5784de009/lib/feed/updaters/sensor_updater.dart#L78) the relevant context (pre-migration but functionally similar). Maybe it's caused by not calling `discoverServices()` when reconnecting, since `existingServices` are found

I'll have to come back to this migration in a few months

chipweinberger, 2023-08-09:
yes I think there might be a regression for Assigned Numbers i.e. 0x180a

chipweinberger, 2023-08-09:
I've looked into this more, and you need to call `discoverServices` after disconnection. This is what Android wants.

closed 2023-08-09 by chipweinberger