chipweinberger/flutter_blue_plus #487: [Question]: Print image

kyawthet-wam, 2023-08-09:
### Ask your question

I want to know this package support printing image. I try the following way.
 ByteData imageData = await rootBundle.load('assets/images/b.png');
          Uint8List dataBytes = imageData.buffer.asUint8List();

          for (int i = 0; i < dataBytes.length; i += 20) {
            final chunk = dataBytes.sublist(
                i, i + 20 < dataBytes.length ? i + 20 : dataBytes.length);
            await characteristic.write(chunk);
          }

But the length of byte list is so big and printing unkown charaters.

chipweinberger, 2023-08-09:
Yes you could use this package to print an image. 

There is also https://pub.dev/packages/blue_print_pos

But you should change its pubspec.yaml to use flutter_blue_plus 1.4.0

closed 2023-08-09 by chipweinberger

kyawthet-wam, 2023-08-09:
@chipweinberger Thanks for reply. Can you please check which part of my code is wrong ?

chipweinberger, 2023-08-09:
it looks correct.

perhaps there is a bug elsewhere

utkuvrs, 2024-07-02:
Hello, this code doesn't work.

It prints out random characters instead.