chipweinberger/flutter_blue_plus #488: [Question]: i am creating send function and that function is given below and also i provide you  output , all is clear but when that send message is not showing in hterm?

NewAge23, 2023-08-09:
### Ask your question

step 1 . this is my send command 
  Future<String> _sendCommand() async {
    String response = '';
    final math = Random();
    List<int> commandData = [
      math.nextInt(256),
      math.nextInt(256),
      math.nextInt(256),
      math.nextInt(256),
    ];
    const int maxChunkSize = 20;

    try {
      List<BluetoothService> services = await widget.device.discoverServices();
      for (BluetoothService service in services) {
        if (service.uuid == Guid('0000ffe0-0000-1000-8000-00805f9b34fb')) {
          BluetoothCharacteristic? characteristic = service.characteristics
              .firstWhereOrNull((c) =>
                  c.uuid == Guid('0000ffe1-0000-1000-8000-00805f9b34fb'));
          if (characteristic != null) {
            for (int i = 0; i < commandData.length; i += maxChunkSize) {
              final chunk = commandData.sublist(
                  i,
                  i + maxChunkSize > commandData.length
                      ? commandData.length
                      : i + maxChunkSize);
              await characteristic.write(chunk,
                  withoutResponse:
                      characteristic.properties.writeWithoutResponse);
              print('Sent data: $chunk');
            }

            await Future.delayed(
                const Duration(milliseconds: 100)); // Allow time for response
            if (response.isEmpty) {
              print('No response received.');
            }
          }
        }
      }
    } catch (e) {
      print('Error sending command: $e');
      // Handle error here
    }

    print('Received response: $response');
    return response;
  }

.......................................................................................................................

 2. output is    
    
D/[FBP-Android](15095): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android](15095): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(15095): discoverServices() - device: 3C:E4:B0:8C:86:6A
D/BluetoothGatt(15095): onSearchComplete() = Device=3C:E4:B0:8C:86:6A Status=0
D/[FBP-Android](15095): [FBP-Android] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](15095): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](15095): [FBP-Android] onCharacteristicWrite: uuid: 0000f.............00000-000000 status: 0
I/flutter (15095): Sent data: [45, 35, 57, 188]
I/flutter (15095): No response received.
I/flutter (15095): Received response:

3. I want to show that data in my hterm software but that data i did not get it   
 this is a software :--  https://www.der-hammer.info/hterm-082-release.html

NewAge23, 2023-08-09:
now i am using latest version of flutter blue plus, but when  i was using 1.5.2 version it perform well , but now none of the version perform correctly

chipweinberger, 2023-08-09:
does it work in FBP 1.7.0? 1.7.1? etc

NewAge23, 2023-08-09:
yes it worked well but now i am not getting why it did no work?

chipweinberger, 2023-08-09:
I don't know. 

Just use 1.5.2. Do not update.

NewAge23, 2023-08-09:
i tried but i did not know why is not perform in this version also .   When i am used your plugin it is greate but now i do not know why its not perform well is this because of i was updated my flutter?

chipweinberger, 2023-08-09:
https://www.rc-network.de/threads/hm-10-modul-mit-hterm-konfigurieren.11894884/

https://soldered.com/learn/hum-bluetooth-module-hm-10/#:~:text=The%20HM%2D10%20uses%20a,that%20new%20data%20is%20available.

https://learn.adafruit.com/reverse-engineering-a-bluetooth-low-energy-light-bulb/explore-gatt

https://stackoverflow.com/questions/18699251/finding-out-android-bluetooth-le-gatt-profiles

https://www.srishtirobotics.com/component/content/article?id=174:hm10-service-uuid-characteristics-uuid&Itemid=435

https://community.thunkable.com/t/why-cant-i-connect-to-device-using-the-same-uuid-characteristic-on-ios-that-works-on-the-android-app/1945721/4

https://stackoverflow.com/questions/73566579/hm-10-ble-module-connect-to-other-devices

closed 2023-08-09 by chipweinberger