chipweinberger/flutter_blue_plus #491: [Question]: Can't see any Bluetooth iBeacon devices when scanning on iOS.

spehj, 2023-08-10:
### FlutterBluePlus Version

1.12.4

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

15.7.7

### Ask your question

I wrote a flutter app to scan for Bluetooth beacons. The Android app works fine, it finds nearby Beacons (I'm testing it with Estimote's iBeacons). But when using iOS, the results are quite different. The `ScanResult` object from Flutter Blue Plus contains different data for the same devices on iOS and Android. Some devices are not even shown on iOS. 

This is how a SearchResult looks like on Android:
```bash
ScanResult{device: BluetoothDevice{remoteId: DC:10:E7:49:A3:A3, localName: , type: BluetoothDeviceType.le, isDiscoveringServices: false, services: null}, advertisementData: AdvertisementData{localName: , txPowerLevel: null, connectable: false, manufacturerData: {76: [2, 21, 185, 64, 127, 48, 245, 248, 70, 110, 175, 249, 37, 85, 107, 87, 254, 109, 206, 253, 111, 201, 193]}, serviceData: {}, serviceUuids: []}, rssi: -70, timeStamp: 2023-08-10 10:42:48.277310connectionState: BluetoothConnectionState.disconnected}
```

And this is how the SearchResult looks like on iOS:
```bash
ScanResult{device: BluetoothDevice{remoteId: E8523648-E08D-1229-60C5-CE248446465F, localName: Seos, type: BluetoothDeviceType.le, isDiscoveringServices: false, services: null}, advertisementData: AdvertisementData{localName: Seos, txPowerLevel: 0, connectable: true, manufacturerData: {302: [21, 74, 211, 182, 0, 182, 42, 70, 76, 48, 78, 48, 86, 48, 49, 70, 78]}, serviceData: {}, serviceUuids: [00009800-0000-1000-8000-00177A000002]}, rssi: -90, timeStamp: 2023-08-10 10:39:47.177018connectionState: BluetoothConnectionState.disconnected}
```

Here is one way to get only iBeacons from all BLE devices nearby.
```dart
SearchBeacon? convertAdvertisementDataToiBeaconData(ScanResult scanResult) {
  // Key for iBeacon (Estimote) is 76 in decimal
  // Add other keys if necessary
  if (scanResult.advertisementData.manufacturerData.containsKey(Constants.iBeaconKey) &&
      scanResult.advertisementData.manufacturerData[Constants.iBeaconKey]!.length == Constants.iBeaconDataLength &&
      scanResult.advertisementData.manufacturerData[Constants.iBeaconKey]![0] == 2 && // 2 here is type of iBeacon
      scanResult.advertisementData.manufacturerData[Constants.iBeaconKey]![1] == 21) {
    // 21 here is length of iBeacon data without type and lentgh bytes (iBeaconDataLength -2)
    return SearchBeacon.fromScanResult(scanResult, Constants.iBeaconKey);
  } else {
    return null;
  }
}
```

I found out that you can't get MAC addresses on iOS, but how about other data like UUIDs, Major, and Minor? Does anyone else work with Flutter Blue or Flutter Blue Plus and iBeacons? I would like to discuss the iOS part.

### Logs

```shell
No logs.
```

chipweinberger, 2023-08-10:
let m know if you figure it out / if there is a bug

chipweinberger, 2023-08-11:
seems a common problem: https://github.com/PhilipsHue/flutter_reactive_ble/issues/772

spehj, 2023-08-11:
I did some [more research](https://stackoverflow.com/questions/25390484/obtaining-bluetooth-le-scan-response-data-with-ios)  and it looks like we'll need to use some other library to scan for iBeacons with iOS devices.

chipweinberger, 2023-08-11:
Yes looks like iOS requires using CoreLocation

chipweinberger, 2023-08-11:
I've updated the README with this limitation.

closed 2023-08-11 by chipweinberger