chipweinberger/flutter_blue_plus #492: [Bug]: Cannot connect to device after device.disconnect()

project2021-code, 2023-08-11:
### FlutterBluePlus Version

1.14.3

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 8

### What happened?

In FlutterBluePlus 1.13.4, I can connect to device after disconnection.

If I use the latest 1.14.3, I cannot connect to device after disconnection. It keeps loop the following msg on Android Studio Console.

### Logs

```shell
/[FBP-Android](10238): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(10238): getConnectedDevices
D/[FBP-Android](10238): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(10238): getConnectedDevices
```

chipweinberger, 2023-08-11:
can you post all the logs?

project2021-code, 2023-08-11:
Updated the log.

joaogbp, 2023-08-11:
I have this same problem.

chipweinberger, 2023-08-11:
I just tried to reproduce this (Android 12), and I did not have a problem.

It might be limited to older android?

chipweinberger, 2023-08-11:
I can see in the logs, `[FBP-Android] onMethodCall: connect` is never called after disconnection. 

This could be a UI problem in the example app. Try updating to 1.14.4!

joaogbp, 2023-08-11:
I am using **flutter_blue_plus: ^1.14.1** and **Android 7.0**.

That is the error when I try to connect again after disconnecting:

`E/flutter ( 8401): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: connect: (code: 6) Timed out after 5s
E/flutter ( 8401): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:59:7)
E/flutter ( 8401): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter ( 8401): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)`

`I/art     ( 9753): Do partial code cache collection, code=14KB, data=30KB
I/art     ( 9753): After code cache collection, code=14KB, data=30KB
I/art     ( 9753): Increasing code cache capacity to 128KB`

`I/BluetoothLeScanner( 9753): startRegisteration: mLeScanClients={com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@3dbc3cc=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@c759e93} ,callback=com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@3dbc3cc
I/BluetoothLeScanner( 9753): startRegisteration: mLeScanClients={} ,callback=com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@3dbc3cc`

I also try the 1.14.4 but got the error:

`
I/flutter (10990): [FBP] <connect> args: {remote_id: 08:3A:F2:43:ED:2A, auto_connect: 0}
I/flutter (10990): [FBP] <connect> result: null
E/flutter (10990): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: connect: (code: 6) Timed out after 5s
E/flutter (10990): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:59:7)
E/flutter (10990): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter (10990): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)`

chipweinberger, 2023-08-11:
5s timeout is too short.

try 35s

project2021-code, 2023-08-12:
https://github.com/boskokg/flutter_blue_plus/assets/124794512/cb51469c-5afb-48db-b3d8-49d5a7bf0057

If I Hot Reload the App after disconnection, it will show the following log:

```
Performing hot reload...
Syncing files to device SM C9000...
Reloaded 2 of 1163 libraries in 2,443ms (compile: 324 ms, reload: 775 ms, reassemble: 607 ms).
D/[FBP-Android](18488): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](18488): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(18488): STATE_ON
D/BluetoothAdapter(18488): STATE_ON
D/BluetoothLeScanner(18488): Start Scan with callback
D/BluetoothLeScanner(18488): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](18488): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(18488): getConnectedDevices
D/[FBP-Android](18488): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(18488): getConnectedDevices
D/[FBP-Android](18488): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(18488): STATE_ON
D/BluetoothAdapter(18488): STATE_ON
D/BluetoothLeScanner(18488): Stop Scan with callback
D/[FBP-Android](18488): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(18488): getConnectedDevices
D/[FBP-Android](18488): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(18488): getConnectedDevices
```

Also, it will not recover even I switch the Bluetooth On/Off.

chipweinberger, 2023-08-12:
Thanks for sharing the video :)

It looks like you are making a custom app based on the example app?

Does the stock example app have this problem?

project2021-code, 2023-08-12:
I am not sure my app is based on the example app. My app use another project (ancient flutter_blue) as a background.

Do you have any guideline of the example app? It seems that it does not include GATT function.

project2021-code, 2023-08-12:
I may find the problem.

In v1.13.4, the `await device.disconnect();` cannot be called twice, while in v1.14.x, the `await device.disconnect();` can be called twice.
The second `device.disconnect()` may cause the library fail to connect after disconnection.

In **example/main.dart**:
Add one to three `await device.disconnect();` after Line 404 and
Comment `return "$prefix ${e.description}";` in Line 600

In **example/pubspec.yaml**"
Add version to Line 16 `flutter_blue_plus: 1.14.7`
Comment `#    path: ../` in Line 22 

```
D/[FBP-Android](27305): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](27305): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](27305): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothLeScanner(27305): could not find callback wrapper
D/[FBP-Android](27305): [FBP-Android] onMethodCall: getAdapterState
D/libGLESv2(27305): STS_GLApi : DTS, ODTC are not allowed for Package : com.boskokg.flutter_blue_plus_example
D/[FBP-Android](27305): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(27305): getConnectedDevices
D/[FBP-Android](27305): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothLeScanner(27305): Start Scan with callback
D/BluetoothLeScanner(27305): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](27305): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(27305): getConnectedDevices
D/[FBP-Android](27305): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](27305): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(27305): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter(27305): isSecureModeEnabled
D/BluetoothGatt(27305): registerApp()
D/BluetoothGatt(27305): registerApp() - UUID=36bc4507-09ae-4747-97f4-707a702f733a
D/BluetoothGatt(27305): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(27305): onClientConnectionState() - status=0 clientIf=7 device=30:AE:A4:28:2C:CE
D/[FBP-Android](27305): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](27305): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27305): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27305): [FBP-Android] onReadRemoteRssi: rssi: -71 status: 0
D/BluetoothGatt(27305): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(27305): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
**D/[FBP-Android](27305): [FBP-Android] onMethodCall: disconnect**
D/BluetoothGatt(27305): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(27305): onClientConnectionState() - status=0 clientIf=7 device=30:AE:A4:28:2C:CE
D/[FBP-Android](27305): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(27305): close()
D/BluetoothGatt(27305): unregisterApp() - mClientIf=7
**D/[FBP-Android](27305): [FBP-Android] onMethodCall: disconnect**
D/[FBP-Android](27305): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothAdapter(27305): STATE_ON
D/BluetoothLeScanner(27305): Stop Scan with callback
```

chipweinberger, 2023-08-12:
> In v1.13.4, the await device.disconnect(); cannot be called twice, while in v1.14.x, the await device.disconnect(); can be called twice.

Can you explain this more? what do you mean "can be called twice"?

chipweinberger, 2023-08-13:
in 1.14.x, if you call `connect` on an already connected device `connectionState` to emit another value, and same for `disconnect`

in 1.13.x if you call `connect` on an already connected device, it will just do nothing, and same for `disconnect`

perhaps this is causing your app to misbehave.

I should probably undo this change.

project2021-code, 2023-08-13:
v1.13.4
https://github.com/boskokg/flutter_blue_plus/assets/124794512/a0c522bc-e8a4-4b77-a930-68efa3a13aa1
If I call `disconnect` twice, the second `disconnect` does nothing.
In the log, only one `[FBP-Android] onMethodCall: disconnect` exists.

v1.14.7
https://github.com/boskokg/flutter_blue_plus/assets/124794512/dcfc0c61-cbe6-4e7f-980b-23d3536c29cb
If I call `disconnect` twice, the second `disconnect` will make the **CONNECT** unavailable.
In the log, there are two `[FBP-Android] onMethodCall: disconnect`.

joaogbp, 2023-08-13:
I changed the timeout but it continues with the problem, also getting another error:

`I/BluetoothLeScanner(23774): startRegisteration: mLeScanClients={com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@2c7e4e0=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@c18e399} ,callback=com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@2c7e4e0
I/BluetoothLeScanner(23774): startRegisteration: mLeScanClients={} ,callback=com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3@2c7e4e0

I/View    (23774): Key down dispatch to io.flutter.embedding.android.FlutterView{c9dce02 VFE...... .F...... 0,0-720,1184 #1}, event = KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_BACK, scanCode=0, metaState=0, flags=0x48, repeatCount=0, eventTime=4685789, downTime=4685789, deviceId=-1, source=0x101 }
I/View    (23774): Key up dispatch to io.flutter.embedding.android.FlutterView{c9dce02 VFE...... .F...... 0,0-720,1184 #1}, event = KeyEvent { action=ACTION_UP, keyCode=KEYCODE_BACK, scanCode=0, metaState=0, flags=0x248, repeatCount=0, eventTime=4685831, downTime=4685789, deviceId=-1, source=0x101 }

I/art     (23774): Do partial code cache collection, code=14KB, data=30KB
I/art     (23774): After code cache collection, code=14KB, data=30KB
I/art     (23774): Increasing code cache capacity to 128KB

E/flutter (23774): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: connect: (code: 6) Timed out after 35s
E/flutter (23774): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:59:7)
E/flutter (23774): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter (23774): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)`

chipweinberger, 2023-08-13:
why does connect become unavailable? 

are you able to add some more print statements and debug breakpoints to analyze what is happening? where does connect get stuck?

chipweinberger, 2023-08-13:
@joaogbp please open a new issue.

Are you able to connect using other BLE scanner apps?

Riyanp, 2023-08-14:
> why does connect become unavailable?
> 
> are you able to add some more print statements and debug breakpoints to analyze what is happening? where does connect get stuck?

I'm also having the same issue after updating to v1.14.7

when i tried to re-connect after disconnected, the `connect()` function does nothing, even the `bluetoothDevice.connectionState.listen` stream suddenly return disconnected state, even when it called multiple times. The only way it can connect again is by hit restart the app.

somehow @project2021-code workaround is working for me, making sure the disconnect doesn't called twice

> v1.14.7
> https://github.com/boskokg/flutter_blue_plus/assets/124794512/dcfc0c61-cbe6-4e7f-980b-23d3536c29cb
> If I call disconnect twice, the second disconnect will make the CONNECT unavailable.
> In the log, there are two [FBP-Android] onMethodCall: disconnect.

project2021-code, 2023-08-14:
Not sure if this help
```
D/[FBP-Android](28051): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](28051): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](28051): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothLeScanner(28051): could not find callback wrapper
D/[FBP-Android](28051): [FBP-Android] onMethodCall: getAdapterState
D/libGLESv2(28051): STS_GLApi : DTS, ODTC are not allowed for Package : com.boskokg.flutter_blue_plus_example
D/[FBP-Android](28051): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(28051): getConnectedDevices
D/[FBP-Android](28051): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothLeScanner(28051): Start Scan with callback
D/BluetoothLeScanner(28051): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](28051): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(28051): getConnectedDevices
D/[FBP-Android](28051): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](28051): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(28051): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter(28051): isSecureModeEnabled
D/BluetoothGatt(28051): registerApp()
D/BluetoothGatt(28051): registerApp() - UUID=fd77ae4a-72f5-487e-bb21-9a49daf46b5d
D/BluetoothGatt(28051): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(28051): onClientConnectionState() - status=0 clientIf=7 device=30:AE:A4:28:2C:CE
D/[FBP-Android](28051): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](28051): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(28051): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](28051): [FBP-Android] onReadRemoteRssi: rssi: -60 status: 0
D/BluetoothGatt(28051): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(28051): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
I/flutter (28051): start 1st disconnect
I/flutter (28051): BluetoothConnectionState.connected
D/[FBP-Android](28051): [FBP-Android] onMethodCall: disconnect
D/BluetoothGatt(28051): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(28051): onClientConnectionState() - status=0 clientIf=7 device=30:AE:A4:28:2C:CE
D/[FBP-Android](28051): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(28051): close()
D/BluetoothGatt(28051): unregisterApp() - mClientIf=7
I/flutter (28051): start 2nd disconnect
I/flutter (28051): BluetoothConnectionState.connected
D/[FBP-Android](28051): [FBP-Android] onMethodCall: disconnect
D/[FBP-Android](28051): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothAdapter(28051): STATE_ON
D/BluetoothLeScanner(28051): Stop Scan with callback
```

example/lib/main.dart Line 404
```
                        print("start 1st disconnect");
                        print(snapshot.data);
                        await device.disconnect();
                        print("start 2nd disconnect");
                        print(snapshot.data);
                        await device.disconnect();
                        print("start 3rs disconnect");
                        print(snapshot.data);
                        await device.disconnect();
                        final snackBar = snackBarGood("Disconnect: Success");
                        snackBarKeyC.currentState?.removeCurrentSnackBar();
                        snackBarKeyC.currentState?.showSnackBar(snackBar);
```

It seems that it stopped at the second `await device.disconnect();`.

joaogbp, 2023-08-14:
Hey @chipweinberger the v1.14.7 now solved my problem. Now I can connect after disconnect without need to rebuild the App.

Thanks for your attention @chipweinberger!

joaogbp, 2023-08-14:
Ops! I just got the error again, now I can connect 2 times (connect and disconnect) but then i can not connect again.

chipweinberger, 2023-08-14:
@joaogbp , can you try master branch? I am not reproducing the problem on android.

To test, I changed the example app to call connect and disconnect twice in a row like this: 

```
await.connect(...)
await.connect(...)
```

and

```
await.disconnect();
await.disconnect();
```

chipweinberger, 2023-08-15:
@joaogbp once you report back i will release 1.14.8

project2021-code, 2023-08-15:
@chipweinberger I can confirm that the problem is solved in the master branch.

I have also tested in my app, and I can connect/disconnect the ESP32.

In the example app log:
```
D/ViewRootImpl@4afdc60[MainActivity]( 6852): ViewPostIme pointer 0
D/ViewRootImpl@4afdc60[MainActivity]( 6852): ViewPostIme pointer 1
I/flutter ( 6852): BluetoothConnectionState.connected
I/flutter ( 6852): start 1st disconnect
D/[FBP-Android]( 6852): [FBP-Android] onMethodCall: disconnect
D/BluetoothGatt( 6852): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt( 6852): onClientConnectionState() - status=0 clientIf=7 device=30:AE:A4:28:2C:CE
D/[FBP-Android]( 6852): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt( 6852): close()
D/BluetoothGatt( 6852): unregisterApp() - mClientIf=7
I/flutter ( 6852): start 2nd disconnect
D/[FBP-Android]( 6852): [FBP-Android] onMethodCall: disconnect
D/[FBP-Android]( 6852): [FBP-Android] already disconnected
I/flutter ( 6852): start 3rd disconnect
D/[FBP-Android]( 6852): [FBP-Android] onMethodCall: disconnect
D/[FBP-Android]( 6852): [FBP-Android] already disconnected
D/[FBP-Android]( 6852): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 6852): STATE_ON
I/chatty  ( 6852): uid=10176(u0_a176) com.boskokg.flutter_blue_plus_example identical 2 lines
D/BluetoothAdapter( 6852): STATE_ON
D/BluetoothLeScanner( 6852): Stop Scan with callback
```

IgoSilvari, 2024-01-25:
@chipweinberger  This error still persists, after disconnecting, it is not possible to connect again, on any other device, and can only be connected after restarting.  

Log

D/[FBP-Android](14060): [FBP] onMethodCall: connect
D/BluetoothGatt(14060): connect() - device: 48:E7:29:A3:88:62, auto: false, eattSupport: false
D/BluetoothGatt(14060): registerApp()
D/BluetoothGatt(14060): registerApp() - UUID=51686bd4-a6f3-4c43-87f7-ca36eb66f2c9
D/BluetoothGatt(14060): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(14060): onClientConnectionState() - status=255 clientIf=10 device=48:E7:29:A3:88:62
D/[FBP-Android](14060): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](14060): [FBP]   status: UNKNOWN_HCI_ERROR (255)
D/BluetoothGatt(14060): close()
D/BluetoothGatt(14060): unregisterApp() - mClientIf=10
I/flutter (14060): FlutterBluePlusException | connect | android-code: 255 | UNKNOWN_HCI_ERROR (255)
D/[FBP-Android](14060): [FBP] onMethodCall: disconnect
D/[FBP-Android](14060): [FBP] already disconnected

chipweinberger, 2024-01-25:
I don't have this problem on my app or my device.

Given the error `UNKNOWN_HCI_ERROR`, your device is malfunctioning.

IgoSilvari, 2024-01-25:
@chipweinberger  I tested on 3 different devices, with different android versions, tested with android 10, 11 and 13

chipweinberger, 2024-01-25:
try different FBP versions?

but tbh, I don't think this is a problem with FBP

IgoSilvari, 2024-01-25:
I tested version 1.31.8, but as I was giving this error, I updated to the latest version 1.31.10, I tested it on 3 devices as I said, but it was from the same brand as Xiaomi, the only difference from the error returned, in the Android versions was that on Android 10, the error code was 133, on Android 13, it was 255, as you can see, in my first message.
​

dontaskit28, 2024-08-28:
I am experiencing the same issue, is there any update on this?

hezhipengzipp, 2025-01-22:
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(requestMtu, device is disconnected, null, null)
                                                                                                    #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
                                                                                                    #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
                                                                                                    <asynchronous suspension>
                                                                                                    #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:630:13)
                                                                                                    <asynchronous suspension>
                                                                                                    #3      BluetoothDevice.requestMtu (package:flutter_blue_plus/src/bluetooth_device.dart:470:7)
                                                                                                    <asynchronous suspension>
                                                                                                    #4      BluetoothDevice.connect (package:flutter_blue_plus/src/bluetooth_device.dart:177:7)
                                                                                                    <asynchronous suspension>

hezhipengzipp, 2025-01-22:
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discoverServices, device is disconnected, null, null)
                                                                                                    #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
                                                                                                    #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
                                                                                                    <asynchronous suspension>
                                                                                                    #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:630:13)
                                                                                                    <asynchronous suspension>
                                                                                                    #3      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:271:7)
                                                                                                    <asynchronous suspension>
                                                                                                    #4      OtaUpgradeManager.discoverServices (package:flutter_lib_parasols/manager/ota_upgrade_manager.dart:103:40)
                                                                                                    <asynchronous suspension>