chipweinberger/flutter_blue_plus #493: I/flutter ( 7149): Exception: PlatformException(write_characteristic_error, data longer than mtu allows. dataLength: 23> max: 20, null, null) #0

faiyaz-shaikh, 2023-08-11:
### FlutterBluePlus Version

 1.9.5

### Flutter Version

3.10.0

### What OS?

Android, iOS

### OS Version

android 12, iPadOS 16.6

### What happened?

Getting an Exception:

**iOS Error**
flutter: Exception: PlatformException(writeCharacteristic, data is longer than MTU allows. dataLen: 23 > maxDataLen: 20, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>
#2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:372:7)
<asynchronous suspension>
#3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:125:5)
<asynchronous suspension>
flutter: 
#0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:165:7)
#1      _RootZone.runBinary (dart:async/zone.dart:1666:54)
#2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
#3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
#4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
#5      Future._completeError (dart:async/future_impl.dart:592:5)
#6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
#7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
flutter: ----------------------------------------------------

**Android Error:**
I/flutter (16474): PlatformException(set_notification_error, gattServer.writeDescriptor() returned false, null, null)
I/flutter (16474): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
I/flutter (16474): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
I/flutter (16474): <asynchronous suspension>
I/flutter (16474): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:194:5)
I/flutter (16474): <asynchronous suspension>
I/flutter (16474): #3      _DexScreenState._readNotification (package:chq_flutter/dex/device_screen.dart:362:5)
I/flutter (16474): <asynchronous suspension>
I/flutter (16474): #4      _DexScreenState.getServices (package:chq_flutter/dex/device_screen.dart:93:5)
I/flutter (16474): <asynchronous suspension>
I/flutter (16474): #5      _DexScreenState.startScan.<anonymous closure> (package:chq_flutter/dex/device_screen.dart:65:11)
I/flutter (16474): <asynchronous suspension>

### Logs

```shell
[FBP-iOS] centralManagerDidUpdateState 5
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: disconnect
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] handleMethodCall: getConnectionState
[FBP-iOS] didConnectPeripheral
[FBP-iOS] handleMethodCall: getConnectionState
[FBP-iOS] handleMethodCall: discoverServices
[FBP-iOS] didDiscoverServices
[FBP-iOS] Found service: 180A
[FBP-iOS] Found service: FFF0
[FBP-iOS] Found service: FFE0
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] handleMethodCall: setNotification
[FBP-iOS] didUpdateNotificationStateForCharacteristic
flutter: Sending line break - [0]
flutter: Received Data!!!! []
flutter: Current State: States.beginning
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
flutter: Sending ENQ - [5]
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] didUpdateValueForCharacteristic 64AB9970-710C-9F44-8FF6-71BF9D900883
flutter: Received Data!!!! [16, 48, 16, 48]
flutter: Current State: States.waitForAckBeginHandshake1AsMaster
flutter: Got ACK! - sending data [16, 1, 67, 84, 79, 48, 48, 49, 48, 48, 48, 49, 82, 82, 48, 49, 76, 48, 49, 16, 3, 134, 37]
[FBP-iOS] handleMethodCall: writeCharacteristic
flutter: ----------------FIREBASE CRASHLYTICS----------------
flutter: Exception: PlatformException(writeCharacteristic, data is longer than MTU allows. dataLen: 23 > maxDataLen: 20, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>
#2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:372:7)
<asynchronous suspension>
#3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:125:5)
<asynchronous suspension>
flutter: 
#0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:165:7)
#1      _RootZone.runBinary (dart:async/zone.dart:1666:54)
#2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
#3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
#4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
#5      Future._completeError (dart:async/future_impl.dart:592:5)
#6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
#7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
flutter: ----------------------------------------------------
```

project2021-code, 2023-08-11:
Do `await widget.device.requestMtu(185);` work on iOS device?

In my memory (Dart 2 flutter_blue_plus v1.5.2), iOS device does not need to use `await widget.device.requestMtu(185);` for adjusting the MTU.

chipweinberger, 2023-08-11:
this has been fixed in recent versions

make sure you listen to the mtu stream on iO

you cannot requestMtu on ios

chipweinberger, 2023-08-11:
closing because this is expected behavior.

closed 2023-08-11 by chipweinberger