chipweinberger/flutter_blue_plus #495: [Bug]: BluetoothConnectionState.disconnected not received

agozie, 2023-08-12:
### FlutterBluePlus Version

1.14.6

### Flutter Version

3.10.6

### What OS?

macOS

### OS Version

13.4.1

### What happened?

I upgraded from 1.12.8 to 1.14.6 and `BluetoothConnectionState.disconnected` stopped being received when the BLE device is switched off, but `BluetoothConnectionState.connected` is received when device connects. So in the code below, `logger.i('****Device disconnected****');` is never called after the update. Please how do I listen for disconnections? Thanks.

`      _deviceStateSub = connectedDevice?.connectionState.listen((state) async {
        logger.i("---------bls: $state");
        if (state == BluetoothConnectionState.connected) {
          // Device is connected
          logger.i('****Device connected****');
        } else if (state == BluetoothConnectionState.disconnected) {
          // Device is disconnected
          logger.i('****Device disconnected****');
        }
      });`

### Logs

```shell
2023-08-12 05:39:37.371 ble_test[33699:3226481] [FBP-iOS] didDisconnectPeripheral: [Error] The connection has timed out unexpectedly.
```

leelai, 2023-08-12:
i think there is a bug in the commit, 
https://github.com/boskokg/flutter_blue_plus/commit/06511c068b603f93b95a9c74a93e40d07af186d7#r124309109

chipweinberger, 2023-08-12:
@leelai yes you are right there is a bug there

thanks for finding it

the bug i see is in the android code though, and this says macos

chipweinberger, 2023-08-12:
@agozie , can you turn on verbose logs? is OnConnectionStateChanged called?

agozie, 2023-08-12:
@chipweinberger, here are the logs after disconnection:

`flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 1, service_data: null, local_name: BG 7C47D4734, tx_power_level: null}, device: {type: 2, remote_id: 3116F046-6A89-357A-4F9B-CFC41BAE6E1C, local_name: BG 7C47D4734}, rssi: -81}}
2023-08-12 12:53:30.882 ble_power_monitor[89150:3587610] [FBP-iOS] didDisconnectPeripheral: [Error] The connection has timed out unexpectedly.
flutter: [FBP] [[ OnConnectionStateChanged ]] result: {remote_id: 604B0829-F569-62B7-064E-1BD0437D6CBE, connection_state: 0, disconnect_reason_code: The connection has timed out unexpectedly., disconnect_reason_string: 6}
2023-08-12 12:53:31.062 ble_power_monitor[89150:3587610] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: {76: 1005331c7bff5c0100000010000000000000000000000000}, service_uuids: null, connectable: 1, service_data: null, local_name: null, tx_power_level: 12}, device: {type: 2, remote_id: 11E1BA37-6EC9-4237-9B90-B2C48858792F, local_name: iPhone 13 Pro Max}, rssi: -52}}
2023-08-12 12:53:31.063 ble_power_monitor[89150:3587610] [FBP-iOS] centralManager didDiscoverPeripheral`

keskink, 2023-08-12:
the fix should be like this
line 848         @"disconnect_reason_code":   error ? @(error.code) : [NSNull null],
line 849         @"disconnect_reason_string": error ? [error localizedDescription] : [NSNull null],

line 871         @"disconnect_reason_code":   error ? @(error.code) : [NSNull null],
line 872        @"disconnect_reason_string": error ? [error localizedDescription] : [NSNull null],

chipweinberger, 2023-08-12:
@agozie looks like `OnConnectionStateChanged` is called, so it "should" work.

Perhaps the `BufferStream` I added to the `connectionState` code is not working. You could try reverting that.

chipweinberger, 2023-08-15:
1.14.8 is released. please open a new issue w/ new logs if you still have problems.

closed 2023-08-15 by chipweinberger