chipweinberger/flutter_blue_plus #497: [Bug]: Android BLUETOOTH_CONNECT runtime permission is required in addition to BLUETOOTH_SCAN

ekuleshov, 2023-08-13:
### FlutterBluePlus Version

1.14.7

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 13

### What happened?

Not sure when this changed, as it used to work in the older version of `flutter_blue_plus`...

Basically triggering BLE scan and then `getRemoteName` on Android 12 or 13, is crashing the app with the `java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission`.

The BLUETOOTH_CONNECT permission needs to be added along with BLUETOOTH_SCAN permission here:

https://github.com/boskokg/flutter_blue_plus/blob/2774fb4dc7e12d0f9a2af59580fd42c814b9741a/android/src/main/java/com/boskokg/flutter_blue_plus/FlutterBluePlusPlugin.java#L329-L334

### Logs

```shell
E/AndroidRuntime(20838): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10370, packageName = ***, attributionTag = null, token = android.os.BinderProxy@92c4b0c, next = null }: AdapterService getRemoteName
E/AndroidRuntime(20838): 	at com.android.bluetooth.Utils.checkPermissionForDataDelivery(Utils.java:916)
E/AndroidRuntime(20838): 	at com.android.bluetooth.Utils.checkConnectPermissionForDataDelivery(Utils.java:948)
E/AndroidRuntime(20838): 	at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.getRemoteName(AdapterService.java:3399)
E/AndroidRuntime(20838): 	at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.getRemoteName(AdapterService.java:3389)
E/AndroidRuntime(20838): 	at android.bluetooth.IBluetooth$Stub.onTransact(IBluetooth.java:1186)
E/AndroidRuntime(20838): 	at android.os.Binder.execTransactInternal(Binder.java:1321)
E/AndroidRuntime(20838): 	at android.os.Binder.execTransact(Binder.java:1280)
```

chipweinberger, 2023-08-14:
will fix.

i removed CONNECT thinking it was not needed during scan.

why does getRemoteName require connect? weird.

chipweinberger, 2023-08-14:
How do you call `getRemoteName`? I don't see `getRemoteName` in our code.

ekuleshov, 2023-08-14:
It is called by Android code to get name of blue device from the scan results.

chipweinberger, 2023-08-14:
thanks for fast response. 

Where in the code do we call that? What FlutterBluePlus code is triggering it? FlutterBluePlus.startScan?

 I just tried on Android 12 and did not get an exception.

chipweinberger, 2023-08-14:
perhaps here?

String localName = scanRecord.getDeviceName();

ekuleshov, 2023-08-14:
Probably.

I'm testing on Android 13 and this issue also been triggered in the original flutter_blue package an in several older forks.

chipweinberger, 2023-08-14:
I'm on android 13 as well. Can you share full logs?

ekuleshov, 2023-08-14:
The above is the only log I get.

Adding back permission resolves issue. The same permission I have to use in my other native Android apps.

chipweinberger, 2023-08-14:
I wanted to see what other FlutterBluePlus functions you are calling in the logs.

> Adding back permission resolves issue. The same permission I have to use in my other native Android apps.

To your manifest? or do you mean in the FBP Java runtime code?

ekuleshov, 2023-08-14:
The error is triggered on the scan call and the app crashes with security exception. Looks like it is unhandled.

The permission is in the app manifest but that is a runtime permission and has to be requested. I had to fork flutter_blue_plus code to add it back.

chipweinberger, 2023-08-14:
Got it.

 I'm just trying to figure out why it happens on your phone and not mine. I am on Android 13 as well.

Perhaps we can solve this issue without required CONNECT in order to SCAN.

chipweinberger, 2023-08-14:
Could you try commenting out this line?

String localName = scanRecord.getDeviceName();

ekuleshov, 2023-08-14:
I do need that, as well as service uuids from the scan results in order to identify supported devices. Perhaps that is where connect comes from.

ekuleshov, 2023-08-14:
Haven't tried. Not sure what part you want me to comment. I was referring to my app functionality.

chipweinberger, 2023-08-14:
In FlutterBluePlusPlugin.java, please comment out this =)

`            1780: String localName = scanRecord.getDeviceName();`

And if that doesn't work, try commenting out more code until you can determine what line in the java is causing this security exception.

ekuleshov, 2023-08-14:
I don't see what else I can comment out to make app not crash without connection permission... That permission is required.

Though you may need to uninstall your test app from device to see fresh permissions.

Also, unlike example app, I'm using the scan() method with subscription for results.

chipweinberger, 2023-08-14:
Thanks for trying. I have not been able to replicate the crash on Android 12. 

I'll likely add it back anyway.

chipweinberger, 2023-08-15:
merged 5b7d1840be117f3783bc196a1a7c664477276825

ekuleshov, 2023-08-15:
> Thanks for trying. I have not been able to replicate the crash on Android 13.

Have you uninstalled your test app?

chipweinberger, 2023-08-15:
No I did not, good point! I'll try that.

chipweinberger, 2023-08-15:
fixed in 1.14.8

MrCsabaToth, 2023-08-16:
Oh, I worked around this by looking at the example, and it was using permission_handler to ask for the connect permission. But I'll drop permission_handler if it's not needed, I didn't spot in the FBP code that particular missing permission, and I thought my issue is with my Android 13 phone. https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/454