chipweinberger/flutter_blue_plus #498: Is there a way to get "disconnect reason" when connection breaks?

shpasser, 2023-08-14:
### FlutterBluePlus Version

N/A

### Flutter Version

N/A

### What OS?

All

### OS Version

N/A

### Ask your question

Is it possible to get the "disconnect reason" when connection breaks? 

### Logs

```shell
No logs
```

shpasser, 2023-08-14:
Could you please answer the question for version 1.5.2 and the latest version as well?

chipweinberger, 2023-08-14:
currently no, we don't expose this.

I would love to add it. Can you think of a good name for the variable? connectionStateDetailed?  connectionStateExtra? connectionStateB?  connectionState2?

shpasser, 2023-08-14:
Since while connecting we do not need the reason I would go with `disconnectReason` for `disconnected` state only. Probably it is a good idea to supply both the integer code and the associated string if possible. Does it make sense?

chipweinberger, 2023-08-14:
merged: https://github.com/boskokg/flutter_blue_plus/commit/0b189ddc4cc01efdcf1682684302278f5aacd49d