chipweinberger/flutter_blue_plus #499: [Bug]: The Nordic UART Service  Can't be found by our API

zhou5852, 2023-08-15:
### FlutterBluePlus Version

^1.14.7

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 9

### What happened?

The Service can't be found by our api 
[image]

### Logs

```shell
I/flutter (17317): service0BluetoothService{remoteId: EF:CC:D1:A2:30:E9, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: EFA2:30:E9, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a00-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: EF:CC:D1:A2:30:E9, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a01-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: fal
I/flutter (17317): service1 BluetoothService{remoteId: EF:CC:D1:A2:30:E9, serviceUuid: 00001801-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [], includedServices: []}
```

chipweinberger, 2023-08-15:
This could be a bug in android, or a bug with your device. Try calling `clearGattCache`.

chipweinberger, 2023-08-17:
did you figure it out?

try using other BLE scanner apps

zhou5852, 2023-08-17:
Oh  thanks for everyone，the problem be solved，thanks too much。 the reason is log is not completed，I can't find the right uuid

closed 2023-08-17 by chipweinberger