chipweinberger/flutter_blue_plus #503: [Question]:

mengzhidaren, 2023-08-18:
### FlutterBluePlus Version

1.4.11

### Flutter Version

3.13.0

### What OS?

Android

### OS Version

android12

### Ask your question

After packaging the apk, the log keeps prompting that the outdated API is used or overwritten. Can you suppress the message in the console?

### Logs

```shell
ע: C:\Users\yyl\AppData\Local\Pub\Cache\hosted\pub.flutter-io.cn\flutter_blue_plus-1.14.11\android\src\main\java\com\boskokg\flutter_blue_plus\FlutterBluePlusPlugin.javaʹ�û򸲸����ѹ�ʱ�� API��
ע: �й���ϸ��Ϣ, ��ʹ�� -Xlint:deprecation ���±��롣
ע: C:\Users\yyl\AppData\Local\Pub\Cache\hosted\pub.flutter-io.cn\flutter_blue_plus-1.14.11\android\src\main\java\com\boskokg\flutter_blue_plus\FlutterBluePlusPlugin.javaʹ����δ�����򲻰�ȫ�Ĳ�����
ע: �й���ϸ��Ϣ, ��ʹ�� -Xlint:unchecked ���±��롣
```

chipweinberger, 2023-08-18:
I don't know what that log is tbh

chipweinberger, 2023-09-11:
we do use some deprecated functions on lower api versions.

not sure if we can do anything about the log without removing support for lower apis.

closed 2023-09-12 by mengzhidaren

chipweinberger, 2023-11-07:
we've marked the deprecated functions in code to prevent these logs