chipweinberger/flutter_blue_plus #505: [Bug]: ^1.13.2 update throttles incoming data, leading to data loss.

agozie, 2023-08-18:
### FlutterBluePlus Version

^1.13.2

### Flutter Version

 3.13.0

### What OS?

iOS

### OS Version

16

### What happened?

I have a hardware that sends sensor readings to my app every 2 seconds. It works perfectly on 1.13.1 and lower. But from 1.13.2, including the new 1.14.11, data is received roughly every 5-6 seconds instead of every 2 seconds and there is a considerable lag in reception. This completely broke parts of the app that requests data from the hardware, 99% of the time, the data is not delivered. This works 100% of the time on 1.13.1 and lower.

Thanks a lot for keeping this lib fresh.

### Logs

```shell
No logs.
```

chipweinberger, 2023-08-18:
We've made no changes to incoming data. Probably some other issue unrelated to the FBP update.

These are the 3 commits in 1.13.2, all unrelated to incoming data: 
https://github.com/boskokg/flutter_blue_plus/commit/e816e31fe624f1ff613037e9db9ddeaa83e7abd5
https://github.com/boskokg/flutter_blue_plus/commit/258794a264bf8771873022348658894390a46578
https://github.com/boskokg/flutter_blue_plus/commit/fe48bd93a4a31b8afac0874458bcb35d6695aad4

agozie, 2023-08-19:
@chipweinberger I tested every recent version multiple times, on both macOS and iOS, same result. v1.13.1 and lower work as expected. v1.13.2 and above are laggy and often fail to receive data from BLE devices.

chipweinberger, 2023-08-20:
very strange.

Can you try testing by moving your pubspec to each commit sequentially?

It would be good to know which commit it causing it.

chipweinberger, 2023-08-24:
Did you figure it out?

agozie, 2023-08-24:
@chipweinberger No. I will test each commit and let you know. Thanks a lot.

agozie, 2023-08-26:
> Thanks for reporting. We've made no changes to incoming data, ever. Try testing more? It might be unrelated to FBP.
> 
> These are the 3 commits in 1.13.2, all unrelated to incoming data: [e816e31](https://github.com/boskokg/flutter_blue_plus/commit/e816e31fe624f1ff613037e9db9ddeaa83e7abd5) [258794a](https://github.com/boskokg/flutter_blue_plus/commit/258794a264bf8771873022348658894390a46578) [fe48bd9](https://github.com/boskokg/flutter_blue_plus/commit/fe48bd93a4a31b8afac0874458bcb35d6695aad4)
> 
> **Note:** _"[Dart] BluetoothDevice: use mutexes to prevent multiple in flight requests"_ only affects `connect`, `disconnect`, `requestMtu`, `discoverServices`, `readRssi`, and `bondState`

I've tested again... same result starting from the first commit.

chipweinberger, 2023-08-26:
from e816e31?

I don't see how that commit would affect anything,

but you can try reverting it.

chipweinberger, 2023-09-02:
any updates. I will close this issue otherwise

agozie, 2023-09-06:
I will test on Android over the weekend and report back. I've been working on other aspects of the app. This issue completely breaks my app's BLE functionality.

agozie, 2023-09-12:
I found that `rssi = await connectedDevice?.readRssi() ?? 0`  in  this function was causing the issue. Removing it solved the problem. Any idea how I can still get the RSSI without causing this issue?

 `readCharacteristic!.setNotifyValue(true);
      readCharacteristicSub =
          readCharacteristic!.lastValueStream.listen((value) async {
        // Handle received data
        try {
          if (value.isNotEmpty) {
            rssi = await connectedDevice?.readRssi() ?? 0;` 

Thanks a lot.

closed 2023-09-12 by agozie

chipweinberger, 2024-01-14:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/754