chipweinberger/flutter_blue_plus #507: [Bug]: writeCharacteristic: (code: 3) Writing is not permitted.

morofir, 2023-08-20:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

15.6

### What happened?

after updating to the newest version of flutter blue plus, im getting writing is not permitted on iOS only.
i was working with 1.3..1 (old) and everythink was fine.
i changed all the deprecated stuff and syntax, but still while trying to write to ble device im getting this errors.

tried to do withoutResponse = true but it didnt help (and i need the response, but was just trying.)

thanks

### Logs

```shell
[FBP-iOS] didWriteValueForCharacteristic: [Error] Writing is not permitted.
flutter: {2023-08-20 13:02:28} {LogLevel.INFO} {BleWrapper} {setCommand} {Command completed: message  (20 bytes) } {trace} {1692525748186}
flutter: {2023-08-20 13:02:28} {LogLevel.INFO} {BleWrapper} {setCommand ERROR} {FlutterBluePlusException: writeCharacteristic: (code: 3) Writing is not permitted.} {trace} {1692525748188}
flutter: {2023-08-20 13:02:28} {LogLevel.INFO} {CommandsManager} {contactCheck} {failed:  (20 bytes) ]} {trace} {1692525748191}
flutter: {2023-08-20 13:02:29} {LogLevel.INFO} {BleWrapper} {setCommand} {Command completed: message [ (20 bytes) ]} {trace} {1692525749204}
flutter: {2023-08-20 13:02:29} {LogLevel.INFO} {CommandsManager} {contactCheck} {success} {trace} {1692525749205}
flutter: {2023-08-20 13:02:29} {LogLevel.INFO} {LocalDbManager} {_handleTreatmentStatusLog} {Succeed insert status log to db MessageType: DeviceLogType.TREATMENT_STATUS_LOG_CURRENT, Timestamp: 2023-08-20 10:01:11.000.} {trace} {1692525749713}
```

chipweinberger, 2023-08-20:
this error comes from iOS. 

are you sure it ever actually worked? it means your characteristic does not support writing.

chipweinberger, 2023-08-24:
Did you figure it out?

chipweinberger, 2023-09-02:
closing. more info needed.

closed 2023-09-02 by chipweinberger