chipweinberger/flutter_blue_plus #508: [Bug]:

expertdev-designer, 2023-08-21:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

Android 11

### What happened?

Got a Bluetooth adapter error while running the app 

### Logs

```shell
D/[FBP-Android]( 8413): [FBP-Android] initializing BluetoothAdapter
D/BluetoothAdapter( 8413): java.lang.Throwable
D/BluetoothAdapter( 8413): 	at android.bluetooth.BluetoothAdapter.<init>(BluetoothAdapter.java:762)
D/BluetoothAdapter( 8413): 	at android.bluetooth.BluetoothAdapter.getDefaultAdapter(BluetoothAdapter.java:735)
D/BluetoothAdapter( 8413): 	at android.bluetooth.BluetoothManager.<init>(BluetoothManager.java:72)
D/BluetoothAdapter( 8413): 	at android.app.SystemServiceRegistry$10.createService(SystemServiceRegistry.java:325)
D/BluetoothAdapter( 8413): 	at android.app.SystemServiceRegistry$10.createService(SystemServiceRegistry.java:322)
D/BluetoothAdapter( 8413): 	at android.app.SystemServiceRegistry$CachedServiceFetcher.getService(SystemServiceRegistry.java:1722)
D/BluetoothAdapter( 8413): 	at android.app.SystemServiceRegistry.getSystemService(SystemServiceRegistry.java:1402)
D/BluetoothAdapter( 8413): 	at android.app.ContextImpl.getSystemService(ContextImpl.java:1940)
D/BluetoothAdapter( 8413): 	at android.content.ContextWrapper.getSystemService(ContextWrapper.java:803)
D/BluetoothAdapter( 8413): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:203)
D/BluetoothAdapter( 8413): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
D/BluetoothAdapter( 8413): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
D/BluetoothAdapter( 8413): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
```

chipweinberger, 2023-08-21:
> =unable to access list of devices

I don't see this log in your logs?

chipweinberger, 2023-08-24:
Did you figure it out?

expertdev-designer, 2023-08-24:
Hii @chipweinberger 
Thank you for your quick response.
Sorry for the late reply.

The error was solved but I could get the device list(mobile device, airdrop with Bluetooth on )
Log: 
`Performing hot restart...
Syncing files to device EB2101...
Restarted application in 2,452ms.
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 9023): isLeEnabled(): ON
D/BluetoothLeScanner( 9023): could not find callback wrapper
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: isAvailable
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter( 9023): isLeEnabled(): ON
D/BluetoothLeScanner( 9023): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 9023): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 9023): isLeEnabled(): ON
E/BLASTBufferQueue( 9023): BLASTBufferItemConsumer::onDisconnect()
E/BLASTBufferQueue( 9023): BLASTBufferItemConsumer::onDisconnect()`

chipweinberger, 2023-08-24:
what was the solution?

expertdev-designer, 2023-08-24:
Hii @chipweinberger 
Not sure what was the issue and it was fixed automatically next day

Now log 
`Performing hot restart...
Syncing files to device EB2101...
Restarted application in 2,609ms.
D/permissions_handler(20658): No permissions found in manifest for: []3
D/permissions_handler(20658): No permissions found in manifest for: []15
D/permissions_handler(20658): Bluetooth permission missing in manifest
D/[FBP-Android](20658): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](20658): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(20658): isLeEnabled(): ON
D/BluetoothLeScanner(20658): could not find callback wrapper
D/[FBP-Android](20658): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](20658): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](20658): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](20658): [FBP-Android] onMethodCall: getAdapterState`

closed 2023-08-31 by chipweinberger