chipweinberger/flutter_blue_plus #509: [Bug]:

umuttsaltan, 2023-08-21:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 11

### What happened?

I'm using setNotifyValue when im getting data from BLE device. When i set this value i get some errors.

 try{ 
          await buttonCharacteristic.setNotifyValue(true); 
         }
         catch(e){
            print("ButtonChar $e");
         }

I used this like that for an exception and im getting these errors : 

```
ButtonChar FlutterBluePlusException: setNotifyValue: (code: 5) notifications were not updated
E/flutter (13773): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: RangeError (index): Invalid value: Valid value range is empty: 0
E/flutter (13773): #0      List.[] (dart:core-patch/growable_array.dart:264)
E/flutter (13773): #1      BleFunctions.buttonChar.<anonymous closure> (package:makelhome2/controllers/ble_functions.dart:116)
E/flutter (13773): #2      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
E/flutter (13773): #3      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
E/flutter (13773): #4      _DelayedData.perform (dart:async/stream_impl.dart:515)
E/flutter (13773): #5      _PendingEvents.handleNext (dart:async/stream_impl.dart:620)
E/flutter (13773): #6      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591)
E/flutter (13773): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40)
E/flutter (13773): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
ButtonChar PlatformException(setNotification, gatt.writeDescriptor() returned false, null, null)
E/flutter (13773): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: RangeError (index): Invalid value: Valid value range is empty: 0
E/flutter (13773): #0      List.[] (dart:core-patch/growable_array.dart:264)
E/flutter (13773): #1      BleFunctions.buttonChar.<anonymous closure> (package:makelhome2/controllers/ble_functions.dart:116)
E/flutter (13773): #2      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
E/flutter (13773): #3      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
E/flutter (13773): #4      _DelayedData.perform (dart:async/stream_impl.dart:515)
E/flutter (13773): #5      _PendingEvents.handleNext (dart:async/stream_impl.dart:620)
E/flutter (13773): #6      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591)
E/flutter (13773): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40)
E/flutter (13773): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
```

### Logs

```shell
/flutter (14956): ButtonChar PlatformException(setNotification, gatt.writeDescriptor() returned false, null, null)
E/flutter (14956): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: RangeError (index): Invalid value: Valid value range is empty: 0
E/flutter (14956): #0      List.[] (dart:core-patch/growable_array.dart:264)
E/flutter (14956): #1      BleFunctions.buttonChar.<anonymous closure> (package:makelhome2/controllers/ble_functions.dart:118)
E/flutter (14956): #2      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
E/flutter (14956): #3      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
E/flutter (14956): #4      _DelayedData.perform (dart:async/stream_impl.dart:515)
E/flutter (14956): #5      _PendingEvents.handleNext (dart:async/stream_impl.dart:620)
E/flutter (14956): #6      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591)
E/flutter (14956): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40)
E/flutter (14956): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
E/flutter (14956): 
I/flutter (14956): [FBP] <discoverServices> result: null
[... 255 lines cut ...]
E/flutter (14956): #6      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591)
E/flutter (14956): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40)
E/flutter (14956): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
E/flutter (14956): 
I/flutter (14956): [FBP] <discoverServices> result: null
I/flutter (14956): [FBP] [[ OnDiscoverServicesResult ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: 84:F7:03:1B:DD:E6, error_code: 0, services: [{included_services: [], characteristics: [{secondary_service_uuid: null, descriptors: [], service_uuid: 00001800-0000-1000-8000-00805f9b34fb, remote_id: 84:F7:03:1B:DD:E6, characteristic_uuid: 00002a00-0000-1000-8000-00805f9b34fb, value: , properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {secondary_service_uuid: null, descriptors: [], service_uuid: 00001800-0000-1000-8000-00805f9b34fb, remote_id: 84:F7:03:1B:DD:E6, characteristic_uuid: 00002a01-0000-1000-8000-00805f9b34fb, value: , properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}], serv
I/flutter (14956): [FBP] <setNotification> args: {remote_id: 84:F7:03:1B:DD:E6, service_uuid: 0000181a-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null, characteristic_uuid: 00002a37-0000-1000-8000-00805f9b34fb, enable: true}
I/flutter (14956): [FBP] <discoverServices> args: 84:F7:03:1B:DD:E6
I/flutter (14956): [FBP] <getConnectedSystemDevices> args: null
I/flutter (14956): ButtonChar PlatformException(setNotification, gatt.writeDescriptor() returned false, null, null)
E/flutter (14956): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: RangeError (index): Invalid value: Valid value range is empty: 0
E/flutter (14956): #0      List.[] (dart:core-patch/growable_array.dart:264)
E/flutter (14956): #1      BleFunctions.buttonChar.<anonymous closure> (package:makelhome2/controllers/ble_functions.dart:118)
E/flutter (14956): #2      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
E/flutter (14956): #3      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
E/flutter (14956): #4      _DelayedData.perform (dart:async/stream_impl.dart:515)
E/flutter (14956): #5      _PendingEvents.handleNext (dart:async/stream_impl.dart:620)
E/flutter (14956): #6      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591)
E/flutter (14956): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40)
E/flutter (14956): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
E/flutter (14956): 
I/flutter (14956): [FBP] <discoverServices> result: null
I/flutter (14956): [FBP] <getConnectedSystemDevices> args: null
I/flutter (14956): [FBP] <getConnectedSystemDevices> result: {devices: [{remote_id: 84:F7:03:1B:DD:E6, type: 2, local_name: PLUG_CHARGER}]}
```

chipweinberger, 2023-08-21:
> PlatformException(setNotification, gatt.writeDescriptor() returned false

this is your issue.

does your characteristic support notifications?

try using the example app!

chipweinberger, 2023-08-26:
did you figure it out?

umuttsaltan, 2023-08-28:
No, i don't know what's the problem. Tried too many things but same error.

Sample function from my code : 

void powerChar() async {
    final powerServiceUuid = Guid("0000181a-0000-1000-8000-00805f9b34fb");
    final powerCharacteristicUuid =
        Guid("7e422156-ab78-11ed-afa1-0242ac120002");

    for (var service in servicesList) {
      if (service.uuid == powerServiceUuid) {
        BluetoothCharacteristic? powerCharacteristic;

        for (var characteristic in service.characteristics) {
          if (characteristic.uuid == powerCharacteristicUuid) {
            powerCharacteristic = characteristic;
            break;
          }
        }

        if (powerCharacteristic == null) {
          print("Power characteristic not found.");
          return;
        }

        try {
          await powerCharacteristic
              .setNotifyValue(powerCharacteristic.isNotifying == false);
        } catch (e) {
          print("Power Char: $e");
        }

        powerCharacteristic.lastValueStream.listen((power) {
          if (power.length == 2) {
            int powerdata = (power[0] << 8) + power[1];

            dataDeviceController.powerData[currentDeviceIndex!] =
                powerdata.toString();
          }
        });
      }
    }
  }

Am i doing something wrong ?

chipweinberger, 2023-08-28:
the problem seems to be your bluetooth device, not your flutter code.

umuttsaltan, 2023-08-31:
I checked the device, descriptors allows only read. Could this be the problem ? Does package trying to writeDesc when setNotify ? If thats the problem can i fix this, change anything on device software ?

chipweinberger, 2023-08-31:
that is a problem. you must change device firmware

chipweinberger, 2023-09-06:
did you fix it?

umuttsaltan, 2023-09-06:
No, my friend works on device firmware. I said the problem, after he change the firmware i'll try it again.

chipweinberger, 2023-09-06:
does it work on iOS?

flierville, 2023-09-07:
Hello, I am joining this conversation because I have encountered the same error message since I upgraded from version 1.4.0 to version 1.14.23

Before with version 1.4.0 it worked with my BLE device (this is not a problem on the device side)
(android, flutter 3.13)

When upgrading to version 1.14.23 the only ones I changed concern the connection (static call, method name etc.)

Thanks for your help

chipweinberger, 2023-09-07:
@flierville , do the notifications work on your device?

if so, you could catch this exception and ignore it.

chipweinberger, 2023-09-07:
@flierville , can you try the FBP example app and see what happens with your device when you read the CCCD?

chipweinberger, 2023-09-08:
So it turns out there was a bug in FBP.

Android actually deprecated the API we were using, because it made this bug easy to make.

In short, `descriptor.getValue()` can only be called after a READ, not a WRITE.

**I've fixed this on master branch. please try it:** dc08c774cd2d1ff4e7f0c8acb6e7221ae38649e7

brain9520, 2023-09-08:
> So it turns out there was a bug in FBP.
> 
> Android actually deprecated the API we were using, because it made this bug easy to make.
> 
> In short, `descriptor.getValue()` can only be reliably called after a READ, not a WRITE.
> 
> **I've fixed this on master branch. please try it:** [dc08c77](https://github.com/boskokg/flutter_blue_plus/commit/dc08c774cd2d1ff4e7f0c8acb6e7221ae38649e7)

I have updated to 1.14.23, but the error is still reported on Android, but it is normal on iOS.

chipweinberger, 2023-09-08:
@brain9520 , you must use master branch

```
dependencies:
  flutter:
    sdk: flutter
  your_package_name:
    git:
      url: https://github.com/boskokg/flutter_blue_plus
      ref: master
```

brain9520, 2023-09-08:
> 

According to what you said, the error is no longer reported, but there are other problems, I will try to solve it myself first, thank you. Then I have another question, that is, flutter_blue_plus does not seem to have duplicate devices when scanning on iOS, but many duplicate Bluetooth devices will appear when scanning on Android. Is there a parameter that can be turned off and set to not repeat scanning.

chipweinberger, 2023-09-08:
>  but there are other problems

what problems?

brain9520, 2023-09-08:
When I use the Android device to start scanning, it will display as follows:
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EC100200
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540

When I scan with an Apple device as follows:
flutter: ble_device_name: EA02001022540
flutter: ble_device_name: midea
flutter: ble_device_name: midea
flutter: ble_device_name: EC100200
flutter: ble_device_name: KeepB3-31
flutter: ble_device_name: HUAWEI Band 7-D5C
flutter: ble_device_name: MITV-2E531
flutter: ble_device_name: HUAWEI Band 7-BFF
flutter: ble_device_name: MITV-900BB

the question is: Is there a parameter to set not to scan devices repeatedly? thank you.

chipweinberger, 2023-09-08:
@brain9520 your issue not not related to notifications, and should not be in this thread

chipweinberger, 2023-09-08:
fixed in 1.14.24

closed 2023-09-08 by chipweinberger

Zqiang12345, 2023-10-27:
I also found this problem. My problem is caused by setting mtu. When setting setNotifyValue, I have to wait for mtu setting to complete before using it