chipweinberger/flutter_blue_plus #513: [Question]:

ringonotts, 2023-08-23:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.1

### What OS?

Android

### OS Version

13

### Ask your question

I am migrating my application from flutter_blue and have the following code for scanning:

```
FlutterBluePlus
            .scan(timeout: const Duration(seconds: 5), androidUsesFineLocation: false)
                .listen((scanResult) {
      // do something with scan result
      _scanListenHandler(scanResult, id);
    }).onDone(() async {
      _isScanning = false;
      await FlutterBluePlus.stopScan();
      setState(() {});
      if (!_isAdding) {
        myUtils.toast("No Sensors detected");
      }
    });
```
I perform the scan and am getting an exception 'another scan already in progress' from flutter_blue_plus.dart line 148.

When i set a breakpoint on line 146 I can see I hit this scan() method twice. The first time I hit the method _isScanning.value == false, and there is a call stack to where i called the scan function.
The second time I hit the breakpoint (note _scanListenHandler() has not been called) _isScanning.value == true and there is no call stack - just a <asyncronous gap>. Obviously it's this second call to scan() causing the exception.

Why is the scan() method called twice ? Am i calling it correctly ? Note I need to know when the scan is complete so i can stop the circular progress indicator in my app (hence not using startScan()).

Thanks in advance, Ringo

### Logs

```shell
I/orthings.batmon(22261): Compiler allocated 4413KB to compile void android.view.ViewRootImpl.performTraversals()
D/CompatibilityChangeReporter(22261): Compat change id reported: 183155436; UID 10263; state: ENABLED
I/FirebaseApp(22261): Device unlocked: initializing all Firebase APIs for app [DEFAULT]
D/CompatibilityChangeReporter(22261): Compat change id reported: 78294732; UID 10263; state: ENABLED
I/flutter (22261): PermissionStatus.granted
D/[FBP-Android](22261): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](22261): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](22261): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](22261): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](22261): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(22261): isLeEnabled(): ON
D/BluetoothLeScanner(22261): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
```

chipweinberger, 2023-08-24:
I don't think I can answer this for you with the code given.

ringonotts, 2023-08-24:
Thanks Chip for getting back to me. Actually you were a great help!

closed 2023-08-24 by ringonotts