chipweinberger/flutter_blue_plus #515: [Bug]: Reading Device Information Service Characteristic fails with "characteristic not found" (short UUIDs)

robert-virkus, 2023-08-25:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

11

### What happened?

I am trying to read the characteristic Serial Number with the UUID `2a23` from the  [Device Information Service](https://www.bluetooth.com/specifications/specs/device-information-service/)  with the UUID `180a`. These UUIDs are defined in the [Bluetooth Assigned Numbers](https://btprodspecificationrefs.blob.core.windows.net/assigned-numbers/Assigned%20Number%20Types/Assigned_Numbers.pdf) document .

With some mapping logic I am able to identify the correct service and characteristic. FBP reports then as 
`0000180a-0000-1000-8000-00805f9b34fb` (for the service `180a`) and `00002a23-0000-1000-8000-00805f9b34fb` for the characteristic `2a23`.

When I try to read the characteristic via `Characteristic.read()`, I receive the following error:
```
[FBP-Android] onMethodCall: readCharacteristic
I/flutter (13116): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (13116): │ PlatformException(readCharacteristic, characteristic not found in service 00002a23-0000-1000-8000-00805f9b34fb, null, null)
I/flutter (13116): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13116): │ #0   StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
message_codecs.dart:652
I/flutter (13116): │ #1   MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
platform_channel.dart:310
I/flutter (13116): │ #2   <asynchronous suspension>
I/flutter (13116): │ #3   FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:302:19)
flutter_blue_plus.dart:302
I/flutter (13116): │ #4   <asynchronous suspension>
I/flutter (13116): │ #5   BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:92:7)
bluetooth_characteristic.dart:92
I/flutter (13116): │ #6   <asynchronous suspension>
```

Expected: I can read the characteristic.

Note that I can read other characteristics from non-standard services just fine. Maybe the assumption that a `Guid` needs to be 16 bytes long is wrong?

### Logs

```shell
Launching lib\main_development.dart on M2101K9G in debug mode...
√  Built build\app\outputs\flutter-apk\app-development-debug.apk.
Connecting to VM Service at ws://127.0.0.1:51797/jh0zWXPcGKQ=/ws
E/LB      (13116): fail to open file: No such file or directory
D/[FBP-Android](13116): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](13116): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](13116): [FBP-Android] onMethodCall: getAdapterState
D/ProfileInstaller(13116): Installing profile for com.dentsplysirona.sirona_cloudconnection.development
I/ion.developmen(13116): ProcessProfilingInfo new_methods=2703 is saved saved_to_disk=1 resolve_classes_delay=8000
D/[FBP-Android](13116): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(13116): isLeEnabled(): ON
D/BluetoothLeScanner(13116): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/[FBP-Android](13116): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(13116): connect() - device: 7C:70:DB:7D:27:19, auto: false
D/BluetoothGatt(13116): registerApp()
D/BluetoothGatt(13116): registerApp() - UUID=303f451d-0bd9-46af-888a-95b2bb3d80fc
D/BluetoothGatt(13116): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(13116): onClientConnectionState() - status=0 clientIf=10 device=7C:70:DB:7D:27:19
D/[FBP-Android](13116): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](13116): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt(13116): configureMTU() - device: 7C:70:DB:7D:27:19 mtu: 512
D/BluetoothGatt(13116): onConnectionUpdated() - Device=7C:70:DB:7D:27:19 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(13116): onConfigureMTU() - Device=7C:70:DB:7D:27:19 mtu=512 status=0
D/[FBP-Android](13116): [FBP-Android] onMtuChanged: mtu: 512 status: 0
D/[FBP-Android](13116): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(13116): discoverServices() - device: 7C:70:DB:7D:27:19
D/BluetoothGatt(13116): onSearchComplete() = Device=7C:70:DB:7D:27:19 Status=0
D/[FBP-Android](13116): [FBP-Android] onServicesDiscovered: count: 6 status: 0
I/flutter (13116): mapping service 0000180a-0000-1000-8000-00805f9b34fb to 180a
I/flutter (13116): mapping device info characteristic from 00002a50-0000-1000-8000-00805f9b34fb to 2a50
I/flutter (13116): mapping device info characteristic from 00002a24-0000-1000-8000-00805f9b34fb to 2a24
I/flutter (13116): mapping device info characteristic from 00002a23-0000-1000-8000-00805f9b34fb to 2a23
I/flutter (13116): mapping device info characteristic from 00002a29-0000-1000-8000-00805f9b34fb to 2a29
D/BluetoothGatt(13116): onConnectionUpdated() - Device=7C:70:DB:7D:27:19 interval=24 latency=0 timeout=500 status=0
I/flutter (13116): service: 00001800-0000-1000-8000-00805f9b34fb: (00002a00-0000-1000-8000-00805f9b34fb, 00002a01-0000-1000-8000-00805f9b34fb, 00002aa6-0000-1000-8000-00805f9b34fb)}
I/flutter (13116): service: 00001801-0000-1000-8000-00805f9b34fb: (00002a05-0000-1000-8000-00805f9b34fb, 00002b29-0000-1000-8000-00805f9b34fb, 00002b2a-0000-1000-8000-00805f9b34fb, 00002b3a-0000-1000-8000-00805f9b34fb)}
I/flutter (13116): service: 180a: (2a24, 2a23, 2a29)}
I/flutter (13116): service: bf5a0000-ec20-424e-a7e3-1c8024e4ede1: (bf5a0001-ec20-424e-a7e3-1c8024e4ede1, bf5a0002-ec20-424e-a7e3-1c8024e4ede1, bf5a0003-ec20-424e-a7e3-1c8024e4ede1, ..., bf5a00ff-ec20-424e-a7e3-1c8024e4ede1, bf5affff-ec20-424e-a7e3-1c8024e4ede1)}
I/flutter (13116): service: bf6a0000-ec20-424e-a7e3-1c8024e4ede1: (bf6a0001-ec20-424e-a7e3-1c8024e4ede1, bf6a0002-ec20-424e-a7e3-1c8024e4ede1, bf6a0003-ec20-424e-a7e3-1c8024e4ede1, ..., bf6a0006-ec20-424e-a7e3-1c8024e4ede1, bf6affff-ec20-424e-a7e3-1c8024e4ede1)}
D/[FBP-Android](13116): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](13116): [FBP-Android] onCharacteristicWrite: uuid: bf5a00ff-ec20-424e-a7e3-1c8024e4ede1 status: 0
D/[FBP-Android](13116): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](13116): [FBP-Android] onCharacteristicRead: uuid: bf5a0001-ec20-424e-a7e3-1c8024e4ede1 status: 0
D/[FBP-Android](13116): [FBP-Android] onMethodCall: readCharacteristic
I/flutter (13116): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (13116): │ PlatformException(readCharacteristic, characteristic not found in service 00002a23-0000-1000-8000-00805f9b34fb, null, null)
I/flutter (13116): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13116): │ #0   StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
I/flutter (13116): │ #1   MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
I/flutter (13116): │ #2   <asynchronous suspension>
I/flutter (13116): │ #3   FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:302:19)
I/flutter (13116): │ #4   <asynchronous suspension>
I/flutter (13116): │ #5   BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:92:7)
I/flutter (13116): │ #6   <asynchronous suspension>
I/flutter (13116): │ #7   BluetoothService._readCharacteristic (package:my_app_name/src/bluetooth/bluetooth_service.dart:384:17)
I/flutter (13116): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13116): │ 09:46:53.468 (+0:00:38.619837)
I/flutter (13116): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13116): │ ⛔ Unable to read characteristic 2a23 (180a)
I/flutter (13116): └───────────────────────────────────────────────────────────────────────────────────────────────────────
Application finished.
Exited (sigterm)
```

chipweinberger, 2023-08-25:
are you sure your device actually has that characteristic?

chipweinberger, 2023-08-25:
feel free to submit a pull request if you fix it properly

chipweinberger, 2023-08-31:
did you figure it out?

closed 2023-09-06 by chipweinberger